Lora-net/LoRaMac-node #751: Battery ADC calculation problem

outely, 2019-06-13:
Hi!
At src/boards/SKiM880B/board.c,  Why is the **batteryVoltage** calculated like this？Is it a error?
```
uint16_t BoardBatteryMeasureVolage(void)
{
    uint16_t vref = 0;
    uint32_t batteryVoltage = 0;

    // Read the current Voltage
    vref = AdcReadChannel(&Adc, ADC_CHANNEL_1);

    // We don't use the VREF from calibValues here.
    // calculate the Voltage in millivolt
    batteryVoltage = (uint32_t)ADC_VREF_BANDGAP * (uint32_t)ADC_MAX_VALUE;
    batteryVoltage = batteryVoltage / (uint32_t)vref;

    return batteryVoltage;
}
```
I think the batteryVoltage calculation should be 
`batteryVoltage = (uint32_t)vref * (uint32_t)ADC_VREF_BANDGAP / (uint32_t)ADC_MAX_VALUE;`

trick2011, 2019-06-13:
It seems alright though. This is found inside the reference manual for the STM32L0x2 on page 322

[image]

As opposed to the next page:
[image]

I do see a optimization oppertunity as the `ADC_VREF_BANDGAP` and `ADC_MAX_VALUE` are constant and could be used in the initialization of `batteryVoltage`.

mluis1, 2019-06-17:
@trick2011 Thanks for providing the answer.

We close this issue.

closed 2019-06-17 by mluis1