Lora-net/LoRaMac-node #755: LoRaMacMcpsRequest gets stuck with LORAMAC_STATUS_BUSY

ChrisEAlfred, 2019-06-20:
We are using the Nucleo-L073 with SX1262MB2xAS on AS923 in Class C.
We on the latest develop git SHA: 6231a5d65806142a6f887e6183227b8da9e49238.

After about 2-4 hours and transmitting every 60 seconds, LoRaMacMcpsRequest is returning LORAMAC_STATUS_BUSY.

It appears that the previous and all other transmissions had a corresponding MCPS indication. However, the last successful transmit does not have a corresponding MCPS indication even after 60 seconds. From there on, all LoRaMacMcpsRequest calls return LORAMAC_STATUS_BUSY.

I placed a breakpoint in LoRaMacMcpsRequest and have confirmed that in LoRaMacIsBusy(), MacCtx.MacState = LORAMAC_TX_RUNNING
and MacCtx.AllowRequests = LORAMAC_REQUEST_HANDLING_ON, hence busy.

The MCPS request is:
mcpsRequest
	Type		unconfirmed
	fPort		3
	fBuffer		20001564
	fBufferSize	240
	Datarate	0

The MacCtx is:
-		MacCtx	{DevEui = 0x00000000, JoinEui = 0x00000000, ...}	LoRaMacCtx_t
		DevEui	0x00000000	uint8_t *
		JoinEui	0x00000000	uint8_t *
		PktBufferLen	253	uint16_t
+		PktBuffer		uint8_t [255]
+		TxMsg	{Type = LORAMAC_MSG_TYPE_DATA, Message = {...}}	LoRaMacMessage_t
+		AppData		uint8_t [255]
		AppDataSize	240 'ð'	uint8_t
+		RxPayload		uint8_t [255]
+		LastTxSysTime	{Seconds = 3245, SubSeconds = 386}	SysTime_t
		MacState	2	uint32_t
+		MacPrimitives	0x20001430 <_ZL4_app+88> {0x0801f091 <McpsConfirm(McpsConfirm_t*)>}	LoRaMacPrimitives_t *
+		MacCallbacks	0x20001440 <_ZL4_app+104> {0x0801f139 <BatteryLevel()>}	LoRaMacCallback_t *
+		RadioEvents	{TxDone = 0x0800a319 <OnRadioTxDone>, ...}	RadioEvents_t
+		TxDelayedTimer	{Timestamp = 0, ReloadValue = 0, IsStarted = false, ...}	TimerEvent_t
+		RxWindowTimer1	{Timestamp = 1475, ReloadValue = 1002, ...}	TimerEvent_t
+		RxWindowTimer2	{Timestamp = 999, ReloadValue = 2040, ...}	TimerEvent_t
		RxWindow1Delay	979	uint32_t
		RxWindow2Delay	1993	uint32_t
+		RxWindow1Config	{Channel = 2 '\002', Datarate = 5 '\005', ...}	RxConfigParams_t
+		RxWindow2Config	{Channel = 0 '\000', Datarate = 2 '\002', ...}	RxConfigParams_t
+		RxWindowCConfig	{Channel = 0 '\000', Datarate = 0 '\000', ...}	RxConfigParams_t
		AdrAckLimit	64	uint16_t
		AdrAckDelay	32	uint16_t
+		AckTimeoutTimer	{Timestamp = 3406, ReloadValue = 4408, ...}	TimerEvent_t
		ChannelsNbTransCounter	1 '\001'	uint8_t
		AckTimeoutRetries	1 '\001'	uint8_t
		AckTimeoutRetriesCounter	1 '\001'	uint8_t
		AckTimeoutRetry	false	_Bool
		NodeAckRequested	false	_Bool
		Channel	2 '\002'	uint8_t
		TxTimeOnAir	395	TimerTime_t
+		McpsIndication	{McpsIndication = MCPS_UNCONFIRMED, ...}	McpsIndication_t
+		McpsConfirm	{McpsRequest = MCPS_UNCONFIRMED, ...}	McpsConfirm_t
+		MlmeConfirm	{MlmeRequest = MLME_JOIN, ...}	MlmeConfirm_t
+		MlmeIndication	{MlmeIndication = MLME_JOIN, ...}	MlmeIndication_t
		RxSlot	RX_SLOT_NONE	LoRaMacRxSlot_t
+		MacFlags	{Value = 1 '\001', Bits = {...}}	LoRaMacFlags_t
		AllowRequests	LORAMAC_REQUEST_HANDLING_ON	LoRaMacRequestHandling_t
+		NvmCtx	0x20000820 <NvmMacCtx> {LORAMAC_REGION_AS923}	LoRaMacNvmCtx_t *

Chris Alfred

mluis1, 2019-07-02:
Thanks for the detailed report.

Unfortunately, we are not able to reproduce the issue using the ClassC example and transmitting an uplink every 30 seconds. Our tests ran successfully for about one week.

Was this report generated using the provided ClassC example and one of the supported platforms?
If it was not the case could you provide a way to reproduce the issue using the ClassC example?

iotfreaks, 2019-07-03:
Hi
We have the same issue and probably we found a problem.  If Rx error occurs during reception of packet in RX1, the bit MacCtx.MacFlags.Bits.MacDone is never set to 1. So, after that, the stack stucks in LORAMAC_TX_RUNNING state because LoRaMacHandleMcpsRequest() is not called inside LoRaMacProcess() function. Please, review function HandleRadioRxErrorTimeout.

iotfreaks, 2019-07-04:
Hi. We make a night-test and problem is gone. This patch solves the problem for us:
```
static void HandleRadioRxErrorTimeout( LoRaMacEventInfoStatus_t rx1EventInfoStatus, LoRaMacEventInfoStatus_t rx2EventInfoStatus )
{
    bool classBRx = false;

    if( MacCtx.NvmCtx->DeviceClass != CLASS_C )
    {
        Radio.Sleep( );
    }

    if( LoRaMacClassBIsBeaconExpected( ) == true )
    {
        LoRaMacClassBSetBeaconState( BEACON_STATE_TIMEOUT );
        LoRaMacClassBBeaconTimerEvent( NULL );
        classBRx = true;
    }
    if( MacCtx.NvmCtx->DeviceClass == CLASS_B )
    {
        if( LoRaMacClassBIsPingExpected( ) == true )
        {
            LoRaMacClassBSetPingSlotState( PINGSLOT_STATE_CALC_PING_OFFSET );
            LoRaMacClassBPingSlotTimerEvent( NULL );
            classBRx = true;
        }
        if( LoRaMacClassBIsMulticastExpected( ) == true )
        {
            LoRaMacClassBSetMulticastSlotState( PINGSLOT_STATE_CALC_PING_OFFSET );
            LoRaMacClassBMulticastSlotTimerEvent( NULL );
            classBRx = true;
        }
    }
    /** ---------------------------------------Added-------------------------------*/
    if(rx1EventInfoStatus == LORAMAC_EVENT_INFO_STATUS_RX1_ERROR)  
    {                                                                                                                
      MacCtx.MacFlags.Bits.MacDone = 1;                                                       
    }    
    /**-------------------------------------------------------------------------------*/                                                                                                              
    if( classBRx == false )
    {
        if( MacCtx.RxSlot == RX_SLOT_WIN_1 )
        {
            if( MacCtx.NodeAckRequested == true )
            {
                MacCtx.McpsConfirm.Status = rx1EventInfoStatus;
            }
            LoRaMacConfirmQueueSetStatusCmn( rx1EventInfoStatus );
        }
        else
        {
            if( MacCtx.NodeAckRequested == true )
            {
                MacCtx.McpsConfirm.Status = rx2EventInfoStatus;
            }
            LoRaMacConfirmQueueSetStatusCmn( rx2EventInfoStatus );

            if( MacCtx.NvmCtx->DeviceClass != CLASS_C )
            {
                MacCtx.MacFlags.Bits.MacDone = 1;
            }
        }
    }

    UpdateRxSlotIdleState( );
}
```

djaeckle, 2019-07-04:
HI @iotfreaks,

thanks for the report and the analysis. At the moment, we are not 100% sure if it is the same issue that @ChrisEAlfred reported. Thus, could you please open up a new issue with a short description of your issue and the configuration you are using?

djaeckle, 2019-07-04:
> Thanks for the detailed report.
>
> Unfortunately, we are not able to reproduce the issue using the ClassC example and transmitting an uplink every 30 seconds. Our tests ran successfully for about one [...]

@ChrisEAlfred, could you provide some more information here? 

> The MCPS request is:
> mcpsRequest
> Type	unconfirmed
> fPort	3
> fBuffer	20001564
> fBufferSize	240
> Datarate	0

Could you explain your transmission policy? It's DR0 with a size of 240 which doesn't fit into the payload for AS923.

ChrisEAlfred, 2019-07-22:
I have just come back from holidays.

I tried @iotfreaks patch and it did not fix the problem.

At the time I reported "fBufferSize 240" the device has trained to DR5.
I will catch another case and report more.

ChrisEAlfred, 2019-07-31:
The problem was setting UsbIsConnected=false thus using battery mode.
Opening a new issue with the correct symptom.

closed 2019-07-31 by ChrisEAlfred

pfaco, 2020-12-23:
I couldn't find a new issue with iotfreaks's problem. But I noticed that his solution is not merged in the master branch, at least.
Could anyone clarify if his problem is solved in a different manner or it simply wasn't reported?