Lora-net/LoRaMac-node #756: LoRaMac-node hangs on SF12 CNF RX2

trick2011, 2019-06-25:
We have had a problem where if a confirmed uplink was sent on SF12 (49 bytes) and a downlink ack was sent in RX2 the stack hangs. This has been confirmed in `LoRa Mac 4.4.1` and ` LoRa Mac 4.4.2-rc.5`. (We are using ST's implementations `V1.1.5` and `V1.2.1`)

Our solution was to check the state of `LoRaMacState`. It was defined as
```
/*!
 * LoRaMac internal states
 */
enum eLoRaMacState
{
    LORAMAC_IDLE          = 0x00000000,
    LORAMAC_TX_RUNNING    = 0x00000001,
    LORAMAC_RX            = 0x00000002,
    LORAMAC_ACK_REQ       = 0x00000004,
    LORAMAC_ACK_RETRY     = 0x00000008,
    LORAMAC_TX_DELAYED    = 0x00000010,
    LORAMAC_TX_CONFIG     = 0x00000020,
    LORAMAC_RX_ABORT      = 0x00000040,
};

/*!
 * LoRaMac internal state
 */
uint32_t LoRaMacState = LORAMAC_IDLE;
```
We determine if it is is staying in a blocked state by comparing it to `0` . If it stays blocked for too long we hard reset our device. (`NVIC_SystemReset`).

However `LoRaMacState` disappeared (at least from) ` LoRa Mac 4.4.2-rc.5` and as such it isn't possible to use this hack anymore.

The underlying problem seems to be an too small of a value for the start/end of the RX2 window.  When SF12 is used with a big message.

Sidenote we are using EU868.

trick2011, 2019-06-25:
Possible relation to https://github.com/Lora-net/LoRaMac-node/issues/755#issue-458281891

mluis1, 2019-07-01:
If I'm not mistaken I guess that the below pull-request solved similar issue.
https://github.com/Lora-net/LoRaMac-node/pull/605 

This pull-request has been merged after the 4.4.2-rc.5 tag. Could you please try it and let us know if it solves your issue.

Could you please also check if the current develop branch presents the same issue?

Please note, that Rx1 and Rx2 windows opening times are related to the end of the transmission. It means that the uplink length has no impact on when they are open. Without the pull-request proposed solution what could happen was that the Tx timed out before the end of the transmission due to the transmission time on air close to the limit of 3 seconds.

trick2011, 2019-07-02:
Thanks for your response. I must've missed that one as I have been searching for a while. We'll be testing this patch from now on and we're expecting this to be the correct solution as it fixes the situation we also determined to be the source.

About the checking of the current develop branch: we're using the ST implementation of this repo and as such much of our code isn't easily upgrade to the newest version but we'll try to test it. (It might take a while as I'll be taking a vacation soon.)

Also thanks for your elaboration about the rx windows.

mluis1, 2019-07-02:
Thanks for the feedback.

For the time being I close this issue. Please feel free to re-open it if your issue is still not solved.

closed 2019-07-02 by mluis1