Lora-net/LoRaMac-node #762: SAML21 platform SX1276 "Wing Board"

KaDw, 2019-07-08:
Hi,

SAML21 port uses SAML21 Xplained Pro with SX1276 "Wing Board". Unforunately SX1276 "Wing Board" is not available in any shop. This setup is a bit out of date and Microchip replaced it with SAMR34. I understand that I can port it myself (eg. to SAMR34) but with up to date hardware more people can contribute and give some feedback. 

Are there any plans to support other hardware platform from Microchip that is not discontinued?

mluis1, 2019-07-19:
Unfortunately, it is not on our plans to add more platforms to this project at short or long term.

closed 2019-07-19 by mluis1

elsalahy, 2020-03-04:
@KaDw we added support for the SAMR34 instead of SAML21
see:
https://github.com/TheThingsNetwork/LoRaMac-node/tree/feature/SAMR34%2BATECC608A

as part of 
https://github.com/TheThingsNetwork/lorawan-example-atecc608a-tnglora

Hope this is still relevant for you.