Lora-net/LoRaMac-node #763: How to enable crypto? (question)

Arcanum7, 2019-07-19:
I have stm32L073 nucleo and sx1272mb2 shield.
I have done my main task to make class A end-node  water-counter  device. It works!
but i don't understand how to make crypto my messages between endnode and base station?
Can somebody link me to example, please.

manojmanu041, 2019-07-30:
Hi,
 https://lorawan-packet-decoder-0ta6puiniaut.runkit.sh/?data=EjRWeQ%3D%3D&nwkskey=6051106ede59074312baafe497ed6ea9&appskey=2dc8afda9d4f52ed877a35f47a1c8a51

U can use think online decoder
Put Nwk skey and app skey .
Put encode data u will get 

One help for I am also using same controller and lora module can u share u r code I am not able to connect lora nod to gateway.

Thanks
Manoj

djaeckle, 2019-08-15:
Hi @Arcanum7,

i don't understand the question. Could you please provide some more information?

manojmanu041, 2019-08-15:
Hi use mqtt protocol .u will get decod value

mluis1, 2019-10-25:
No activity, We close the issue.

closed 2019-10-25 by mluis1