Lora-net/LoRaMac-node #767: LoRaMacMcpsRequest gets stuck with LORAMAC_STATUS_BUSY with battery power

ChrisEAlfred, 2019-07-31:
We are testing the example classA on a NucleoL073 with SX1262MB2CAS shield.

If UsbIsConnected is changed from true to false at board.c line 148 (inside BoardInitMcu() to simulate battery power, then MCPS confirm reports "Tx timeout" and then LoRaMacMcpsRequest gets stuck with LORAMAC_STATUS_BUSY.

~~~code
###### ===== ClassA demo application v1.0.RC1 ==== ######

DevEui      : ****************************
AppEui      : ****************************
AppKey      : ***************************************************

###### ===== JOINED ==== ######

ABP

DevAddr     : 000000FF
NwkSKey     : 02 6F 52 D7 44 50 70 BA 25 9E 3F 6C DA FA 64 C3
AppSKey     : 1E 1E 89 2E 4C 61 99 2E 8F 94 36 95 0D 1E 59 B6

###### ===== MCPS-Request ==== ######
STATUS      : OK

###### ===== MCPS-Confirm ==== ######
STATUS      : Tx timeout

###### ===== UPLINK FRAME 1 ==== ######

CLASS       : A

TX PORT     : 2
TX DATA     : UNCONFIRMED
00

DATA RATE   : DR_2
U/L FREQ    : 923200000
TX POWER    : 0
CHANNEL MASK: 0003

###### ===== MCPS-Request ==== ######
STATUS      : OK

###### ===== MCPS-Request ==== ######
STATUS      : Busy

###### ===== MCPS-Request ==== ######
STATUS      : Busy

###### ===== MCPS-Request ==== ######
STATUS      : Busy

###### ===== MCPS-Request ==== ######
STATUS      : Busy

###### ===== MCPS-Request ==== ######
STATUS      : Busy

~~~

ChrisEAlfred, 2019-08-03:
I believe low power is broken.

(I am at latest development git SHA: 32554735fc4e8c764b150497115e71bd67341050)

The only ClassA examples that use low power are SKi881AXL, SKi880B Ski899A. As I am using the NucleoL073, I tried porting the SKi881AXL (also based on L0) board.c and got the same results.

Starting with the unchanged NucleoL073 example I then:
(1) set UsbIsConnected=false in board.c line 148 to enable going to sleep
(2) setup a timer to keep the processor awake for 7 seconds starting from entering DEVICE_STATE_SEND in main.c

As a result of the above, the CPU stays awake for 7 seconds on every transmit.

This appears to be working so far (will leave over night).

My suspicion is that IsMacProcessPending in main.c is not being set to 1 correctly.
~~~code
static uint8_t IsMacProcessPending = 0;
...
void OnMacProcessNotify( void )
{
    IsMacProcessPending = 1;
}

int main( void)
               ...
                if( IsMacProcessPending == 1 )
                {
                    // Clear flag and prevent MCU to go into low power modes.
                    IsMacProcessPending = 0;
                }
                else
                {
                    // The MCU wakes up through events
                    BoardLowPowerHandler( );
                }

~~~

ChrisEAlfred, 2019-08-06:
Using the timer to keep the device awake has been running for over 3 days.

ChrisEAlfred, 2019-08-22:
This is rather important as we can not use low power well in ClassA.
We have a timer running to keep the processors running long enough for the ClassA transaction to complete.

It appears IsMacProcessingPending is not being set to 1 (OnMacProcessNotify not called) in all cases to keep the CPU awake when required.

mluis1, 2019-10-25:
I guess that I have found the issue. I have replicated your setup and so far the uplinks are being done correctly (No "Tx Timeout". I'll let the node run with the fix over the weekend.

Please find below a patch with the change I did:
```patch
diff --git a/src/boards/NucleoL073/cmsis/stm32l0xx_hal_conf.h b/src/boards/NucleoL073/cmsis/stm32l0xx_hal_conf.h
index f331eed9..09dccd78 100644
--- a/src/boards/NucleoL073/cmsis/stm32l0xx_hal_conf.h
+++ b/src/boards/NucleoL073/cmsis/stm32l0xx_hal_conf.h
@@ -150,8 +150,8 @@
 #define  VDD_VALUE                    ((uint32_t)3300U) /*!< Value of VDD in mv */           
 #define  TICK_INT_PRIORITY            ((uint32_t)0U)    /*!< tick interrupt priority */            
 #define  USE_RTOS                     0U     
-#define  PREFETCH_ENABLE              0U              
-#define  PREREAD_ENABLE               1U
+#define  PREFETCH_ENABLE              1U              
+#define  PREREAD_ENABLE               0U
 #define  BUFFER_CACHE_DISABLE         0U
 
 /* ########################## Assert Selection ############################## */
```

mluis1, 2019-10-25:
Also, I supposed that you are using AS923 as region isn't it?

ChrisEAlfred, 2019-10-25:
Yes I am using AS923.
In writing the description I unfortunately left off that critical information.

mluis1, 2019-10-28:
Thanks, I let the end-device run over the weekend and this morning everything looks to operate as expected.

Could you please confirm if the proposed solution solves the issue on your side?
I will then fix this for the different platforms.

mluis1, 2019-11-08:
@ChrisEAlfred Had you some time to verify that the proposed fix Is OK for you?

mluis1, 2019-11-13:
The proposed fix should fix the reported issue.
As there is no feedback we close the issue.

closed 2019-11-13 by mluis1