Lora-net/LoRaMac-node #768: Can't get downlink when using SX126x

Montanari9, 2019-08-06:
Hi,

I've been trying for a while now to run an AT Slave application using the SX1262, it works properly with the SX127x but with the SX1262 I can only transmit correctly.

Here is my hardware setup:
I'm using the B-L072Z-LRWAN1 to validate the SX127x 
I'm using the NUCLEO-G071RB+SX1262MB2CAS (shield)
As for network, I'm using the everynet (ns.atc.everynet.io) and I can see that the TX from my board is always received, but when I schedule a downlink, it only works for the SX127x

Checking and comparing both codes behavior, I could see that whenever I use the command "AT+SEND=2:TEST" , I can see that the flow with the SX127x will have the STM32L072 RTC's ALARM configured several times and it will open the window for reception, but the same command for the SX126x will only schedule once. So, my main suspicion is related to the TimerListHead, I can see that running each code, the content of this struct is changed several times for the SX127 and the same doesn't happen for the SX126x.

Going even further I can see that the SX1276SetSleep function will call the TimerStop, which loads the values for the TimerListHead right in the beginig of the code (under the LORA_Init function) but the same flow for the SX126x will call the SX126xSetSleep, but inside it, there is actually no update for the TimerListHead, it looks like the SX126x would be the one responsible for waking the MCU, so I'm lost here

Thanks and Best Regards

ChrisEAlfred, 2019-08-07:
Is this possibly related to #767?
Try running without sleep. #767 reports that the CPU is prematurely going to sleep.

Montanari9, 2019-08-07:
Hi Chris, 

Thanks for the quick reply, I've actually tried with and without low power, same effect. It does looks like it's something missing in the API to handle the SX126x in the scheduling activities

djaeckle, 2019-08-13:
Hi Montanari9,

which version are you using?

mluis1, 2020-02-21:
Is the issue solved?
Can we close this issue?

mluis1, 2020-03-18:
No feedback, we close the issue.

closed 2020-03-18 by mluis1