Lora-net/LoRaMac-node #770: classC upstrem corrupted data after ABP join

wwx0715, 2019-08-08:
When I test classA ABP routing using NucleoL476, node upload wrong command(many invlid command id). I debug this problem and found LoRaMacCommandsRemoveCmd function normal even if NvmCtx.MacCommandList is NULL and make NvmCtx.SerializedCmdsSize is huge value, then application can not send any data.
I try to solve this problem using bellow code, please confirm this problem.

diff --git a/src/mac/LoRaMac.c b/src/mac/LoRaMac.c
index a5d41775..3f95f55a 100644
--- a/src/mac/LoRaMac.c
+++ b/src/mac/LoRaMac.c
@@ -2081,7 +2081,7 @@ static void ProcessMacCommands( uint8_t *payload, uint8_t macIndex, uint8_t comm
     bool adrBlockFound = false;
     bool adrInvalidBlockFound = false;
     uint8_t macCmdPayload[2] = { 0x00, 0x00 };
-    MacCommand_t* macCmd;
+    MacCommand_t* macCmd = NULL;^M
 
     while( macIndex < commandsSize )
     {
diff --git a/src/mac/LoRaMacCommands.c b/src/mac/LoRaMacCommands.c
index 914ba715..6cd1d1e7 100644
--- a/src/mac/LoRaMacCommands.c
+++ b/src/mac/LoRaMacCommands.c
@@ -168,7 +168,7 @@ static bool LinkedListInit( MacCommandsList_t* list )
  */
 static bool LinkedListAdd( MacCommandsList_t* list, MacCommand_t* element )
 {
-    if( ( list == 0 ) && ( element == 0 ) )
+    if( ( list == 0 ) || ( element == 0 ) )
     {
         return false;
     }
@@ -203,7 +203,7 @@ static bool LinkedListAdd( MacCommandsList_t* list, MacCommand_t* element )
  */
 static MacCommand_t* LinkedListGetPrevious( MacCommandsList_t* list, MacCommand_t* element )
 {
-    if( ( list == 0 ) && ( element == 0 ) )
+    if( ( list == 0 ) || ( element == 0 ) )
     {
         return NULL;
     }
@@ -239,7 +239,7 @@ static MacCommand_t* LinkedListGetPrevious( MacCommandsList_t* list, MacCommand_
  */
 static bool LinkedListRemove( MacCommandsList_t* list, MacCommand_t* element )
 {
-    if( ( list == 0 ) && ( element == 0 ) )
+    if( ( list == 0 ) || (list->First == 0) || ( element == 0 ) )
     {
         return false;
     }

djaeckle, 2019-08-15:
Hi wwx0715,

thanks for the report! I will double check it and will come back to you as soon as possible.

closed 2019-11-13 by mluis1

ouoam, 2020-05-16:
Why line 346 not replace 0 with NULL

https://github.com/Lora-net/LoRaMac-node/blob/823a3fdacbb1ac7bb22512915404d2bb184b957c/src/mac/LoRaMacCommands.c#L345-L350

mluis1, 2020-05-18:
Thanks for the report.
Most certainly because we missed it.