Lora-net/LoRaMac-node #773: ifdef __cplusplus extern "C" { not put in headers

JulianvDoorn, 2019-08-14:
At the moment adding boards inside C++ project is a bit of a hassle in regards to C++ linkage vs C linkage. This is due to 

```
#ifdef __cplusplus
extern "C" {
#endif
```

Currently my workaround is
extern "C" {
#include "board.h"
...
}

Thank you

ChrisEAlfred, 2019-08-15:
I think that's ok, the lora stack is C, not C++.

JulianvDoorn, 2019-08-18:
That's right, however one of the major advantages IMO of using C is to be able to target more potential users (C and C++ users). And C++ isn't backwards compatible for no reason. I think it's nice to not having to worry whether my includes have C or C++ linkage. Granted, it is fairly trivial to add extern "C" around my includes, but I feel like it shouldn't be that way.

Hopefully I'll get around to make a PR adding extern "C" to all header files using a script. But I've found that the include guards are inconsistent across the sources.

trick2011, 2019-08-21:
I support the adding of `extern "C"` and uniforming the include guards.

The addition of `extern C` to the headers is little effort when done consistently and makes the integration of this library inside C++ projects easier without impacting the C users. As a reference: ST's implementation of this library has the addition of these declaration but is by default a C project.

djaeckle, 2019-08-22:
@JulianvDoorn ,

thanks, i see the point.

PR #805 (2019-10-31): Fixed C linkage for C++ projects
Resolves #773 
Wrote a Python script adding the C linkage 'guards'. Please double check whether all headers that are affected by the references issue are solved.
closed 2019-10-31 by mluis1