Lora-net/LoRaMac-node #774: Class C on develop reports "No free channel found"

ChrisEAlfred, 2019-08-19:
We are on the develop branch at 180c8deff3697bb944e8d2aad188aac5899db2f8

I have tested on the NucleoL073 with the SX1276 and SX1262 in Class C, APB and keep getting:
~~~
###### ===== MCPS-Request ==== ######
STATUS      : No free channel found
~~~

The commit at 0bd3b6a498b362be95b6c0d7b925e92776df790e adds the following code t DEVICE_STATE_SEND to set the class after joining as class A:
~~~code
                    mibReq.Type = MIB_DEVICE_CLASS;
                    LoRaMacMibGetRequestConfirm( &mibReq );

                    if( mibReq.Param.Class!= CLASS_C )
                    {
                        mibReq.Param.Class = CLASS_C;
                        LoRaMacMibSetRequestConfirm( &mibReq );
                    }
~~~

Changing the class to Class C ultimately calls SX126xSetOperatingMode( MODE_RX ) via OpenContinuousRxCWindow( ). So the mode is now MODE_RX.

BUT ...

When we return to execute:

~~~code
NextTx = SendFrame( );
~~~

This tries to call Radio.IsChannelFree() which fails due to the mode not being idle.

djaeckle, 2019-08-19:
Hi ChrisEAlfred,

thanks for the report. I will double check it.

djaeckle, 2019-08-20:
Hi @ChrisEAlfred ,

i can confirm the issue for regions using LBT, namely AS923 and KR920 .

djaeckle, 2019-08-22:
Hi,

it will take some time to apply a suitable fix. The problem is, that the LBT function changes the radio state and thus, its configuration. Here, we need to make sure, changing the configuration has no effect on other operations of the MAC, especially, if no free channel is available.

KwonTae-young, 2019-09-05:
I am using `v4.4.2`
I tried to use the KR920 band

I am not able to use the KR920 due to the following message.
```
###### ===== MCPS-Request ==== ######
STATUS      : No free channel found

###### ===== MCPS-Request ==== ######
STATUS      : No free channel found

###### ===== MCPS-Indication ==== ######
STATUS      : Address fail

###### ===== MCPS-Request ==== ######
STATUS      : No free channel found

###### ===== MCPS-Indication ==== ######
STATUS      : Address fail
```

mluis1, 2019-10-23:
The provided fix should solve the initial issue.

Currently, you may disable the functionality by changing the threshold value. You may change it to 0.
Like this the device will think that the channel is always free.

Concerning the the LBT activation for AS923 is due to the fact that Japan is part of the AS923 region. Japan regulatory office mandates the usage of LBT. Thus, if one wants to sell devices using the same firmware on different countries including Japan, then the LBT must be activated.

In the future we will add an MCPS-Request which will allow disabling/enabling the LBT checks. By default we will set it disabled. Then it is up to the application layer to enable/disable the LBT when required.

closed 2019-10-23 by mluis1