Lora-net/LoRaMac-node #776: Forcing back to DR0

ChrisEAlfred, 2019-08-28:
Is it sufficient to just do the following to force the MAC back to DR0?:

~~~code
                mibSet.Type = MIB_CHANNELS_DATARATE;
                mibSet.Param.ChannelsDatarate = DR0;
                LoRaMacMibSetRequestConfirm( &mibSet );
~~~

djaeckle, 2019-08-30:
Hi @ChrisEAlfred,

yes, if the region allows it. However, please note that you need to specify the DR if ADR is off also with every MCPS request.

ChrisEAlfred, 2019-08-31:
Thank you,

For others, the reason for this request is to have a 'test connection' button for user confidence of operation.

You want to make sure the device is going to do its best to connect with the easiest DR.

closed 2019-09-01 by ChrisEAlfred

Pirakutu, 2019-09-02:
If you wanna use it as a test connection you would also wanna probably set the power to max again (the network server might have optimized this also)