Lora-net/LoRaMac-node #777: Common names used (e.g. spi.h) for common headers cause name collisions

JulianvDoorn, 2019-08-29:
I ran into a name collision where the Atmel Software Framework's SPI (asf/sam/drivers/spi/spi.h) driver's header conflicts with the LoRaMac SPI HAL (src/system/spi.h) when porting LoRaMac to another platform. (Both are included via `#include "spi.h"`, even in internal files, which complicates it a lot)

The issue I'm running into is that I cannot include both headers without changing one of two libraries. And both libraries unfortunately expect subdirectories to be included recursively, which is more of a hassle with asf than with LoRaMac, since the latter does it for you via CMake. 

Currently my workaround is renaming one of the two headers and all the files that include it. My preference goes to LoRaMac since asf is enormous and includes spi.h all over the place.

I hope one of the following refactors could be done:
- Prefix all LoRaMac HAL headers with `loramac-hal-` or just `loramac-`
- Prefix all LoRaMac headers with `loramac-` and make it a library-wide convention (it does make stuff more verbose)
- Refactor the entire project so that all includes are done relative from the `include/` (or `src/`, since .c and .h files are mixed) folder

Thank you.

Kamilcuk, 2019-09-30:
I also hope that one day it would also be possible to refactor all Gpio* Spi* Rtc* etc. symbol names with proper namespaces like loramac_Gpio* and loramac_Spi* and loramac_Rtc* etc.  
Alternatively to "prefix all LoRaMac headers with loramac-" - the headers could be also put inside `loramac` folder and then `#include "loramac/spi.h"`.

mluis1, 2020-09-01:
This improvement is on our to do list and we are working on this kind of updates.

If you don't mind we close this issue as it will take some time to get the naming conventions updated across the project.

closed 2020-09-01 by mluis1