Lora-net/LoRaMac-node #779: Node stuck in DEVICE_STATE_SLEEP

Almaz1c, 2019-08-31:
Hi!
We got following on successful JOIN of `SX1272MB2xAS` dev board:

`###### ===== MLME-Request - MLME_JOIN ==== ######
STATUS      : Duty-cycle restricted
###### ===== ClassA demo application v1.0.RC1 ==== ######
DevEui      : xx-xx-xx-xx-xx-xx-xx-xx
AppEui      : xx-xx-xx-xx-xx-xx-xx-xx
AppKey      : xx-xx-xx-xx-xx-xx-xx-xx xx-xx-xx-xx-xx-xx-xx-xx
###### ===== MLME-Request - MLME_JOIN ==== ######
STATUS      : OK
###### ===== JOINING ==== ######
###### ===== MLME-Confirm ==== ######
STATUS      : OK
###### ===== JOINED ==== ######
OTAA
DevAddr     : 01702220
DATA RATE   : DR_0
###### ===== MCPS-Request ==== ######
STATUS      : Duty-cycle restricted
###### ===== MCPS-Request ==== ######
STATUS      : Duty-cycle restricted
###### ===== MCPS-Request ==== ######
STATUS      : Duty-cycle restricted`

There are `SX1272MB2xAS` output in a case of disabled gateway:
`###### ===== MCPS-Request ==== ######
STATUS      : Duty-cycle restricted
###### ===== ClassA demo application v1.0.RC1 ==== ######
DevEui      : xx-xx-xx-xx-xx-xx-xx-xx
AppEui      : xx-xx-xx-xx-xx-xx-xx-xx
AppKey      : xx-xx-xx-xx-xx-xx-xx-xx xx-xx-xx-xx-xx-xx-xx-xx
###### ===== MLME-Request - MLME_JOIN ==== ######
STATUS      : OK
###### ===== JOINING ==== ######
###### ===== MLME-Confirm ==== ######
STATUS      : Rx 2 timeout
###### ===== MLME-Request - MLME_JOIN ==== ######
STATUS      : Duty-cycle restricted`

But here is output of our custom lorawan node no matter gateway exist or not:
`###### Hello world
###### ===== ClassA demo application v1.0.RC1 ==== ######
DevEui      : xx-xx-xx-xx-xx-xx-xx-xx
AppEui      : xx-xx-xx-xx-xx-xx-xx-xx
AppKey      : xx-xx-xx-xx-xx-xx-xx-xx xx-xx-xx-xx-xx-xx-xx-xx
###### ===== MLME-Request - MLME_JOIN ==== ######
STATUS      : OK
###### ===== JOINING ==== ######
`
Debugger shows no any hard fault. Program works but it loops in `while` loop without leaving `DEVICE_STATE_SLEEP` state.
What can be the issue?

Lucianovici, 2019-09-16:
What platform/board are you using?

mluis1, 2019-10-25:
You most certainly have issues with connection between the MCU and the Radio.
You may also need to check if your porting of the board directory is correct.
Please, check that used SPI and DIO pins are the ones connected to the radio.

From the description it looks like that the MCU didn't saw interrupts coming from the radio.

mluis1, 2020-02-21:
Is this issue solved?
Can we close the issue?

mluis1, 2020-03-18:
No feedback, we close the issue.

closed 2020-03-18 by mluis1