Lora-net/LoRaMac-node #783: How to make the SX126X sleep

richard-ubi, 2019-09-05:
HI

I am debugging the SX1261 sleep mode on the board. When I remove the SX1261 from the hardware and the whole board current is 1.2ua, my stm32 sleep mode should be no problem.

So I would like to ask two questions; First: When stm32 sleeps, the IO that connects SX1261 should be set to what mode, I tried a lot, including semtech's IO setting method is as follows, and can not succeed;
void SX126xIoDeInit( void )
{
    GpioInit( &SX126x.Spi.Nss, RADIO_NSS, PIN_ANALOGIC, PIN_PUSH_PULL, PIN_PULL_UP, 1 );
    GpioInit( &SX126x.BUSY, RADIO_BUSY, PIN_ANALOGIC, PIN_PUSH_PULL, PIN_NO_PULL, 0 );
    GpioInit( &SX126x.DIO1, RADIO_DIO_1, PIN_ANALOGIC, PIN_PUSH_PULL, PIN_NO_PULL, 0 );
}

In addition, the way I enter sleep mode is as follows, I don't know if it is correct, please ask my friend for help, thank you very much.

void SX126X_Sleep( void )
{
		RadioStatus_t status;
                SleepParams_t params = { 0 };
		status.Value=0;
		params.Value=0;
		SX126xReset( );
		delay_ms( 20 );
		SX126xSetStandby( STDBY_RC );
		delay_ms( 20 );
		SX126xSetSleep( params )；	
                delay_ms( 1 );
}

Pirakutu, 2019-09-05:
If you keep the sx126x powered, then you should leave the NSS pin set as an output and on HIGH(1). I am guessing the radio busy pin could be set as analog also when the sx126x is powered, this should not have an inpact on the MCU side (i would leave it the way it was initialized first and then try to put it to analog and see the difference), i would do the same with the IRQ pin DIO1 also.

richard-ubi, 2019-09-06:
I am almost the same as you think, but power consumption is still a problem.

Pirakutu, 2019-09-06:
you have the same or almost the same, what is different? on the code you shared i see all pins being deinitialized as analog, which would not be the same.

richard-ubi, 2019-09-07:
OK, thanks！finally found the difference in the hardware design circuit, 1261 and 1262 hardware design is different, I used the SX1261 pin10 connected to pin1, resulting in higher current, but when I disconnected PIN1 and PIN10, When using DC-DC, my code can't connect to ttn through the gateway, still in the test.

richard-ubi, 2019-09-09:
Finally I re-sold the SX1261, hardware problem...

closed 2019-09-09 by richard-ubi