Lora-net/LoRaMac-node #787: Downlink Data Fragment header

KaDw, 2019-09-10:
Hi,

According to LoRaWAN Fragmented Data Block Transport Specification header for Downlink Data Fragment it should be: 
[image]

I think in `LmhpFragmentationOnMcpsIndication` fragCounter is parsed the other way around:
```
                fragCounter = ( mcpsIndication->Buffer[cmdIndex++] << 0 ) & 0x00FF;
                fragCounter |= ( mcpsIndication->Buffer[cmdIndex++] << 8 ) & 0xFF00;

                fragIndex = ( fragCounter >> 14 ) & 0x03;
                fragCounter &= 0x3FFF;
```
whereas it should be:
```
                fragCounter = ( mcpsIndication->Buffer[cmdIndex++] << 8 ) & 0xFF00;
                fragCounter |= ( mcpsIndication->Buffer[cmdIndex++] << 0 ) & 0x00FF;

                fragIndex = ( fragCounter >> 14 ) & 0x03;
                fragCounter &= 0x3FFF;
```
Eg. LoRaMac-node expects 8th data fragment to be `08 09 00` instead of `08 00 09` 
Also, similar bug is present in loraserver.io which generates wrong frames. I assume that loraserver.io is also used for testing fuota so headers are compatible

mluis1, 2019-09-10:
The current LoRaMacNode and I guess loraserver.io implmentations are correct.

If you look at the beginning of the specification document you will notice that multi-byte data is sent LSB first.

[image]

KaDw, 2019-09-10:
Thanks, that makes sense now.

closed 2019-09-10 by mluis1