Lora-net/LoRaMac-node #788: LoRaWan 1.0.3

queifaro, 2019-09-10:


queifaro, 2019-09-10:
Hello,
what about of pre certification test for class B and C ?
Thanks in advance
B/R
Fabio

mluis1, 2019-09-10:
The LoRa Alliance has not yet defined certification tests for Class B and Class C.

Although, we have tested this code against different network server providers which support Class B and Class C.

queifaro, 2019-09-10:
OK Thanks.
Could you indicate the network server you use for class B and C test ?

Also , do you know if LoRa Alliance consider a Class B specification in 1.0.3 final ?
Thanks again
B/R
Fabio

mluis1, 2019-10-25:
The ClassC and ClassB certification will be possible starting at version 1.0.4 and future ones.

We have tested the ClassB and ClassC implementation against different network operators providers.
We have also tested against the open source network server http://loraserver.io

closed 2019-10-25 by mluis1