Lora-net/LoRaMac-node #789: Confirmed message retry even if ack is received

MaheshKumarPN, 2019-09-11:
Hi ,

I have found that the confirmed message is retried ,even if ack is reveiced via downlink message,This is due to  the use of `OnAckTimeoutTimerEvent();` to stop ack timeout timer But it sets retry flag

```
static void OnAckTimeoutTimerEvent( void* context )
{
    PTM_Log("OnAckTimeoutTimerEvent\n");
    TimerStop( &MacCtx.AckTimeoutTimer );

    if( MacCtx.NodeAckRequested == true )
    {
        MacCtx.AckTimeoutRetry = true;
    }
    if( MacCtx.NvmCtx->DeviceClass == CLASS_C )
    {
        MacCtx.MacFlags.Bits.MacDone = 1;
    }
    if( ( MacCtx.MacCallbacks != NULL ) && ( MacCtx.MacCallbacks->MacProcessNotify != NULL ) )
    {
        MacCtx.MacCallbacks->MacProcessNotify( );
    }
}
```
in the following schenario,if     MacCtx.NodeAckRequested= true,MacCtx.McpsConfirm.AckReceived = true.then MacCtx.AckTimeoutRetry is set to true there by causing retry of message.
   ```
 // Verify if we need to disable the AckTimeoutTimer
    if( MacCtx.NodeAckRequested == true )
    {
        if( MacCtx.McpsConfirm.AckReceived == true )
        {
            OnAckTimeoutTimerEvent( NULL );
        }
    }
    else
    {
        if( MacCtx.NvmCtx->DeviceClass == CLASS_C )
        {
            OnAckTimeoutTimerEvent( NULL );
        }
    }
```
I understand that this can fixed by stoping timer and setting  MacCtx.AckTimeoutRetry = false of by an function like this below.
```
static void OnAckReceivedEvent( void* context )
{
    PTM_Log("OnAckReceivedEvent\n");
    TimerStop( &MacCtx.AckTimeoutTimer );

    if( MacCtx.NodeAckRequested == true )
    {
        MacCtx.AckTimeoutRetry = false;
    }
    if( MacCtx.NvmCtx->DeviceClass == CLASS_C )
    {
        MacCtx.MacFlags.Bits.MacDone = 1;
    }
    if( ( MacCtx.MacCallbacks != NULL ) && ( MacCtx.MacCallbacks->MacProcessNotify != NULL ) )
    {
        MacCtx.MacCallbacks->MacProcessNotify( );
    }
}
 ```

Or Do we have  a better approach?
Or Is that something else causing this problem?

djaeckle, 2019-10-22:
HI MaheshKumarPN,

thanks for the report. Which version are you using?

MaheshKumarPN, 2019-12-12:
Sorry for late reply ,it is observed in v4.42

shrikrishnab, 2020-03-02:
Hi MaheshKumarPN,
in your application, in MacProcessNotify(), do you immediately perform call to  "LoRaMacProcess"? If yes, then you see this issue.
Typically ( or what I have done is) MacProcessNotify() should set a flag in application. Once application sees the flag is set, it should then call "LoRaMacProcess". 

MacProcessNotify() can be called multiple times from stack, that's how it is designed (may be expert may have more details on this). LoRa node application has to ensure that it calls "LoRaMacProcess" only once.

GarethFunk, 2020-03-06:
I'm having a similar issue. The fix proposed by @MaheshKumarPN (which to clarify, is a change to `ProcessRadioRxDone()` ) works as far as my testing has shown. It then allows you to run `LoRaMacProcess()` in the callback function.

As far as what @shrikrishnab said, the reference implementation certainly behaves in that way. I think that's the safest way to implement it. That being said, it puts a great deal of constraint on building an application if this is the case.

GarethFunk, 2020-03-06:
Here's what I've come up with for my callback:
```
void LoRaMacProcessNotify()
{
    // Each time we call the function, update the counter
    ++lora_mac_process_count;
    if(lora_mac_process_count == 1)
    {
        // If we are the first caller
        while(lora_mac_process_count > 0)
        {
            // This prevents calls nesting but makes sure they are subsequently processed
            LoRaMacProcess();
            --lora_mac_process_count;
        }
    }
    return;
}
```
This allows the application to be unaware of `LoRaMacProcess()` and prevents the bug caused by allowing recursive calls without modifying the library files. 
In my testing it works as well as the suggestion by @MaheshKumarPN 
I'd be interested to know if anyone can see a drawback.

djaeckle, 2020-03-06:
Hi all,

thanks for the report and all the analysis. 

@MaheshKumarPN , do you call the function `LoRaMacProcess()` inside the callback `LoRaMacProcessNotify()`?

In general, please note that the function `LoRaMacProcess()` **shall not** be called inside a callback function of the LoRaMac. Instead it shall be called outside a the callback functions of the LoRaMac e.g. as it is done in the sample applications.

GarethFunk, 2020-03-06:
@djaeckle From what I can tell, there are reasonable sensitive timing constraints around when the `LoRaMacProcess()` function is called after the `LoRaMacProcessNotify()`. Am I correct about that? Can you give some details about when it needs to be called in terms of both the sequence and the timing restrictions?

MaheshKumarPN, 2020-03-27:
Hi,

Sorry for late reply
In my application
LoRaMacProcess() is directly called inside LoRaMacProcessNotify().
I would suggest to doccument this if possible :)

mluis1, 2020-03-27:
This project is based on a cooperative system.
It means that all lengthy operations should be avoided.

The `LoRaMacProcess` API should be called as often as possible.

The `LoRaMacProcessNotify` callback goal is to notify the application layer that the MAC layer has pending operations to be done.
In the provided examples we just set a boolean variable to represent this state.
In the examples this variable is used to prevent the MCU to go in sleep mode and call the `LoRaMacProcess` API up until there is no more pending operations, which, means the MCU can go in sleep mode.

Sometimes we can avoid lengthy (blocking) operations. Some sensors may require long delays to in order to provide their data.
In such cases we have 2 options:
1. Call the `LoRaMacProcess` API while performing the wait operation.
```c
    TimerTime_t startTime = TimerGetCurrentTime( );
    while( ( TimerGetCurrentTime( ) - startTime ) < LONG_WAIT_TIME_MS )
    {
        LoRaMacProcess( );
    }
```
2. Implement the sensor driver in such way that it does not block the processing.  
The best way to achieve this is to implement a state machine and have an API to query the current sensor state. Thus, for a lengthy operation there would be at least 3 states. `SENSOR_STATE_LENGTHY_OP_START`, `SENSOR_STATE_LENGTHY_OP_WAIT`, `SENSOR_STATE_LENGTHY_OP_STOP`
```c
SensorState_t SensorState = SENSOR_STATE_IDLE;
TimerTime_t StartTime = 0;

SensorState_t SensorProcess( void )
{
    SensorState_t newState = SensorState;

    switch( SensorState )
    {
    case ...: // Represents other sensor states handling
        newState = SENSOR_STATE_????; 
        break;
    case SENSOR_STATE_LENGTHY_OP_START:
        ... // Required operations to start the lengthy wait
        SensorStartTime = TimerGetCurrentTime( );
        newState = SENSOR_STATE_LENGTHY_OP_WAIT; 
        break;
    case SENSOR_STATE_LENGTHY_OP_WAIT:
        if( ( TimerGetCurrentTime( ) - startTime ) > LONG_WAIT_TIME_MS )
        {
            newState = SENSOR_STATE_LENGTHY_OP_STOP;
        }
        break;
    case SENSOR_STATE_LENGTHY_OP_STOP:
        ... // Required operations to stop the lengthy wait
        newState = SENSOR_STATE_????; 
        break;
    }
   SensorState = newState;
   return newState;
}

void main( void )
{
    SensorInit( );
    while( 1 )
    {
        LoRaMacProcess( );
        if( SensorProcess( ) == SENSOR_STATE_DATA_READY )
        {
            SensorDataLoRaMacTx( );
        }
    }
}
```

Hope the explanation is clear enough.

MaheshKumarPN, 2020-03-30:
Cool....Thanks👌

MedadRufus, 2020-03-30:
>    1. Implement the sensor driver in such way that it does not block the

Thanks for the useful example

MedadRufus, 2020-03-30:
I was not very sure how to set things up for a GPS, which takes quite a
while to return a GPS fix value(expecially if it is trying to get a fix
from a cold start).

Thanks

closed 2020-04-03 by mluis1