Lora-net/LoRaMac-node #791: SAML21 with SX1272MB2DAS

Lucianovici, 2019-09-15:
Hi,

I see there is only `sx1276-board.c` implemented for SAML21.
I tried to get inspire from `NucleoL476`'s [sx1272 implementation ](https://github.com/Lora-net/LoRaMac-node/blob/develop/src/boards/NucleoL476/sx1272mb2das-board.c) however I was unable to make it so far.

Can you please guide me to implement sx1272 for SAML21?

Thanks.

Lucianovici, 2019-09-16:
Alright i created a pull request for sx1272 on SAM platform.
https://github.com/Lora-net/LoRaMac-node/pull/792

Please review this.
Thanks.

mluis1, 2019-10-25:
The changes you performed look to be correct.

The only thing that requires some attention concerns the pins setup mainly the interrupts. Unfortunately, we can't check this.

closed 2019-10-25 by mluis1