Lora-net/LoRaMac-node #793: Using USART_3 for GPS RX/TX with SX1276MB1LAS and Nucleo152

slyfoxneil, 2019-09-19:
Hello,

Thanks for open sourcing this loramacnode software project. It has been very helpful . I have been trying to build my own device for the  last 4 weeks and this github repo helps a lot.

Anyway, I was trying to use USART_3 or UART_5 for periodic-uplink-lpp (Nucleo152) with issue of  hanging up after PPS is detected. 

Using USART_1(using PB7/PB6 rx/tx) like that of NAMote72 example code works, but I need to rewire with jumpers wires since PB6 is assigned RADIO_NSS  for  SX1276MB1LAS (as defined n board-config.h). I moved the RADIO_NSS pin to PB13 for the setup to work properly. This setup works great aside from modifying the gps.c routine to parse GNGGA for ublox GPS module.

In attempting to use USART_3 or UART5 revised the uart-board.c to use USART_3/UART_5 same way as I added USART1.  I have defined the correct RX/TX GPIO pins on board-config.h for this cases.(PB11/PB10 for USART3, PC12/PD2 for UART5 - RX/TX pairs).  USART_1 uses PB7/PB6 rx/tx pairs since the default PA10/P9 rx/tx pairs are used for RADIO_DIO_0 and RADIO_DIO_4 respectively by the SX1276MB1LAS board.

All parameters are the same, except for USART3/UART 5 specific definitions like GPIO_AF7_USART3, __HAL_RCC_USART3_FORCE_RESET and so on.

After such changes, and no compile error (I am using KDevelop in Linux), and loading to the mcu, it hangs up on GpsMcuOnPpsSignal() after doing UartConfig().

Any help is greatly appreciated and thanks you in advance.

All the best!

Neil Ablang

mluis1, 2020-03-24:
Can we close this issue?

slyfoxneil, 2020-03-24:
Yes  close

closed 2020-03-24 by mluis1