Lora-net/LoRaMac-node #794: Latest release has NVM context manager, but comment says it is still untested

jeremyherbert, 2019-09-21:
`NvmCtxMgmt.c` contains the following comments:

```
/*!
 * Enables/Disables the context storage management storage at all. Must be enabled for LoRaWAN 1.1.x.
 * WARNING: Still under development and not tested yet.
 */
```

```
/*!
 * Enables/Disables maximum persistent context storage management. All module contexts will be saved on a non-volatile memory.
 * WARNING: Still under development and not tested yet.
 */
```

But the latest release notes imply that the feature is not experimental anymore. It would be good to clarify which of these is true.

mluis1, 2019-10-25:
Yes, the mention is there because we haven't tested it on all supported platforms.

But, at least for NucleoL073 and B-L072Z-LRWAN1 platforms we didn't saw issues yet.

closed 2019-10-25 by mluis1

darkfader, 2020-04-06:
Well, it seems buggy to me. DataBlockAdrCnt keeps on increasing.
Is it supposed to be some circular buffer? ;)