Lora-net/LoRaMac-node #796: length error with no payload in class B

victordemin, 2019-09-25:
I got following length error while trying to come in class B with no payload, only MAC request:
(REGION US915, DR_0)

(18:02:06.368) ###### ===== BEACON 1253458944 ==== ######
(18:02:06.368) GW DESC     : 0
(18:02:06.368) GW INFO     : E3 38 0E E3 38 0E 
(18:02:06.368) 
(18:02:06.368) FREQ        : 923300000
(18:02:06.368) DATA RATE   : DR_8
(18:02:06.368) RX RSSI     : -62
(18:02:06.368) RX SNR      : 8
(18:02:06.368) 
 (18:02:12.429) PrepareFrame: Available payload  : 11
(18:02:12.429) PrepareFrame: macCmdsSize 12
 (18:02:12.429) MAX PAYLOAD, Fopt , lenN   : 11    12    0
(18:02:12.429) 
(18:02:12.429) ###### ===== MCPS-Request ==== ######
(18:02:12.429) STATUS      : Length error
 (18:02:27.374) 
 (18:02:27.435) PrepareFrame: Available payload  : 11
(18:02:27.435) PrepareFrame:  macCmdsSize 12
 (18:02:27.435) MAX PAYLOAD, Fopt , lenN   : 11    12    0
(18:02:27.435) 
(18:02:27.435) ###### ===== MCPS-Request ==== ######
(18:02:27.435) STATUS      : Length error

Fopt length is 12 bytes while max payload is 11 bytes. For such situation there is a function LoRaMacCommandsSerializeCmds which remove extra Fopt bytes. But actual length itr is nowhere updated. May be it must be like this:

LoRaMacCommandStatus_t LoRaMacCommandsSerializeCmds( size_t availableSize, size_t* effectiveSize, uint8_t* buffer )
{
    if( ( buffer == NULL ) || ( effectiveSize == NULL ) )
    {
        return LORAMAC_COMMANDS_ERROR_NPE;
    }
    MacCommand_t* curElement;
    curElement = NvmCtx.MacCommandList.First;
    uint8_t itr = 0;

    // Loop through all elements
    while( curElement != NULL )
    {
        // If the next MAC command still fits into the buffer, add it.
        if( ( availableSize - itr ) >= ( CID_FIELD_SIZE + curElement->PayloadSize ) )
        {
            buffer[itr++] = curElement->CID;
            memcpy1( &buffer[itr], curElement->Payload, curElement->PayloadSize );
            itr = itr + curElement->PayloadSize;
        }
        else
        {
            break;
        }
        curElement = curElement->Next;
    }

  *effectiveSize = itr;
   NvmCtx.SerializedCmdsSize = itr;

    return LORAMAC_COMMANDS_SUCCESS;
}

?

ConnyBusy, 2019-09-25:
Hi @victordemin,
this is a known issue and there is a fix for it which will be available in version 4.4.3

victordemin, 2019-09-26:
Ok. Thank you!

closed 2019-09-26 by victordemin