Lora-net/LoRaMac-node #797: LmhpClockSyncOnMcpsIndication switch

KaDw, 2019-09-26:
Hi,

I think there is an issue with parsing commands in LmhpClockSync.c
```
    // ======= remove this =======
    while( cmdIndex < mcpsIndication->BufferSize )
    {
    // ======= remove this =======
        switch( mcpsIndication->Buffer[cmdIndex++] )
        {
            case CLOCK_SYNC_PKG_VERSION_REQ:
            {
                LmhpClockSyncState.DataBuffer[dataBufferIndex++] = CLOCK_SYNC_PKG_VERSION_ANS;
                LmhpClockSyncState.DataBuffer[dataBufferIndex++] = CLOCK_SYNC_ID;
                LmhpClockSyncState.DataBuffer[dataBufferIndex++] = CLOCK_SYNC_VERSION;
                break;
            }
            case CLOCK_SYNC_APP_TIME_ANS:
            {
                // Check if a more precise time correction has been received.
                // If yes then don't process and ignore this answer.
                // break here
                if( mcpsIndication->DeviceTimeAnsReceived == true )
                {
                    break;
                }
                int32_t timeCorrection = 0;
                timeCorrection  = ( mcpsIndication->Buffer[cmdIndex++] << 0  ) & 0x000000FF;
                timeCorrection += ( mcpsIndication->Buffer[cmdIndex++] << 8  ) & 0x0000FF00;
                timeCorrection += ( mcpsIndication->Buffer[cmdIndex++] << 16 ) & 0x00FF0000;
                timeCorrection += ( mcpsIndication->Buffer[cmdIndex++] << 24 ) & 0xFF000000;
                if( ( mcpsIndication->Buffer[cmdIndex++] & 0x0F ) == LmhpClockSyncState.TimeReqParam.Fields.TokenReq )
                {
                    SysTime_t curTime = { .Seconds = 0, .SubSeconds = 0 };
                    curTime = SysTimeGet( );
                    curTime.Seconds += timeCorrection;
                    SysTimeSet( curTime );
                    LmhpClockSyncState.TimeReqParam.Fields.TokenReq = ( LmhpClockSyncState.TimeReqParam.Fields.TokenReq + 1 ) & 0x0F;
                    if( LmhpClockSyncPackage.OnSysTimeUpdate != NULL )
                    {
                        if( ( timeCorrection <= -1 ) || ( timeCorrection >= 1 ) )
                        {
                            LmhpClockSyncPackage.OnSysTimeUpdate( );
                        }
                    }
                }
                break;
            }
            case CLOCK_SYNC_APP_TIME_PERIOD_REQ:
            {
                // Increment index
                cmdIndex++;
                // TODO implement command prosessing and handling
                LmhpClockSyncState.DataBuffer[dataBufferIndex++] = CLOCK_SYNC_APP_TIME_PERIOD_ANS;
                // Answer status not supported.
                LmhpClockSyncState.DataBuffer[dataBufferIndex++] = 0x01;

                SysTime_t curTime = SysTimeGet( );
                // Substract Unix to Gps epcoh offset. The system time is based on Unix time.
                curTime.Seconds -= UNIX_GPS_EPOCH_OFFSET;
                LmhpClockSyncState.DataBuffer[dataBufferIndex++] = ( curTime.Seconds >> 0  ) & 0xFF;
                LmhpClockSyncState.DataBuffer[dataBufferIndex++] = ( curTime.Seconds >> 8  ) & 0xFF;
                LmhpClockSyncState.DataBuffer[dataBufferIndex++] = ( curTime.Seconds >> 16 ) & 0xFF;
                LmhpClockSyncState.DataBuffer[dataBufferIndex++] = ( curTime.Seconds >> 24 ) & 0xFF;
                break;
            }
            case CLOCK_SYNC_FORCE_RESYNC_REQ:
            {
                // TODO implement command prosessing and handling
                break;
            }
        }
// ======= remove this =======
    }
// ======= remove this =======
```
If we get answer from more precise source (DeviceTimeReq MAC command) we will break from switch:
```
                if( mcpsIndication->DeviceTimeAnsReceived == true )
                {
                    break;
                }
```
Then, cmdIndex will not be increased. Next because of this loop`while( cmdIndex < mcpsIndication->BufferSize )` we will search for command in whole buffer instead of just first byte.

Eg.
We receive downlink on port 202 with payload `01 FA BE 8C 5D 00`, then enter 'CLOCK_SYNC_APP_TIME_ANS' from which we will break without incrementing `cmdIndex` (because time sync came from DeviceTimeReq). Next we will loop through remaining payload `FA BE 8C 5D 00` and on `00` send `CLOCK_SYNC_PKG_VERSION_REQ` that server never asked for. 

I think `while( cmdIndex < mcpsIndication->BufferSize )` should be removed.

KaDw, 2019-10-07:
@djaeckle  @mluis1 can you confirm this issue?

mluis1, 2019-10-25:
Yes, you are right.
But we cannot remove the `while` loop because the specification allows the concatenation of commands.

Our proposition to solve the issue is shown below:
```c
...
                if( mcpsIndication->DeviceTimeAnsReceived == true )
                {
                    cmdIndex += 4;
                    break;
                }
...
```

mluis1, 2019-10-31:
@KaDw, Could you please confirm that the proposed fix is OK?

KaDw, 2019-10-31:
@mluis1 I tested it and it's working fine. Thank you

mluis1, 2019-10-31:
Thanks, for the feedback.

closed 2019-10-31 by mluis1