Lora-net/LoRaMac-node #799: AU915 not receiving downlink in RX2

sergiohernandezmontero, 2019-10-07:
Hello,

I'm using Murata's Type ABZ  with the ST's  I-CUBE-LRWAN 1.2.1, wich includes  LoRaMAC.c v4.4.2-rc.5.

I have succesfully achieved to make EU868 and US915 Class C nodes with multicast work without meaningful problems but when I try to generate an AU915 version I observe that I cannot receive anything in my Rx2 window.

I'm using a Multitech Conduit gateway and LoRaServer 2.6. 

Some tests I have run:

**Test 1:**
Force LoRaServer to always answer JoinAccept and Mac Commands in RX1. => Node joins succesfully and receives Mac commands. Do not receive unicast nor multicast frames  in Rx2 window when node is not emitting.

OnRadioRxDone() is not reached when transmitting downlinks through RX2.

**Test 2:**
Force LoRaServer to always answer JoinAccept and Mac Commands in RX2. => Node does not join network.

OnRadioRxDone() is not reached when transmitting downlinks through RX2.

Any hints?

Thank in advance to everybody.

EduardoPfeifer, 2019-10-07:
Hello!
Please, compare your RegionAU915.h with this https://github.com/Lora-net/LoRaMac-node/blob/develop/src/mac/region/RegionAU915.h

sergiohernandezmontero, 2019-10-08:
Thanks. I've updated RegionAU915.h and now works. Thank you.

mluis1, 2019-10-25:
Can we close the issue?

closed 2020-02-21 by mluis1