Lora-net/LoRaMac-node #800: Timestamp in microsend?

selimsagir, 2019-10-07:
Hello, there is subsecond in the in the rtc-board library. Is there a way for get nanosecond or subsecond timesamp with this library or with another way?  I want to add timestamp to ADC datas for reference. Thanks.

mluis1, 2019-10-25:
This project requires to have a millisecond precision only. 

In order to get bigger precision you must modify the low level timers. Please note that it will require a lot of work to do so and you will loose the low-power feature.

If a millisecond precision is enough you may get the current timestamp by calling `SysTimeGetMcuTime` which will provide the number of seconds and milliseconds elapsed since MCU reset.

In case you have synchronized the system time with the network server you may also use `SysTimeGet` which will provide the current time in seconds and milliseconds.

closed 2019-10-25 by mluis1

selimsagir, 2019-10-25:
I am trying to get microsecond and nanosecond wtih SysTick. For now i am getting values but i need precise and same values for two different kit.

mluis1, 2019-10-25:
Unfortunately, we can't help you. 
These kind of questions are more related to the MCU usage. I would advise that you ask your question on the MCU manufacturer's forum.

Explaining that you would like to get a timestamp in nanoseconds using the SysTick timer.

selimsagir, 2019-10-25:
Thank  you for advice.