Lora-net/LoRaMac-node #801: cmake crashes generating KDevelop3 project

hovercraft-github, 2019-10-11:
Is there a workaround for the following situation?
Cmake crashes generating KDevelop3 project for the "develop" branch of LoRaMac-node.
Other variants generated ok, e.g. "Eclipse CDT4 - Unix Makefiles".
I'm using: Ubuntu 18.04, gcc-arm-none-eabi-8-2019-q3-update, cmake version 3.10.2
The log:
```
cmake -DCMAKE_TOOLCHAIN_FILE="cmake/toolchain-arm-none-eabi.cmake" -GKDevelop3 ..
-- The C compiler identification is GNU 8.3.1
-- The CXX compiler identification is GNU 8.3.1
-- Check for working C compiler: /home/al/local/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-gcc
-- Check for working C compiler: /home/al/local/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/al/local/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /home/al/local/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Linker script: /home/al/prj/LoRa/LoRaMac-node/src/boards/NucleoL073/cmsis/arm-gcc/stm32l073xx_flash.ld
-- The ASM compiler identification is GNU
-- Found assembler: /home/al/local/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-gcc
-- Configuring done
cmake: /build/cmake-t1NJEA/cmake-3.10.2/Source/cmGeneratorTarget.cxx:885: void cmGeneratorTarget::GetSourceFiles(std::vector<std::__cxx11::basic_string<char> >&, const string&) const: Assertion `this->GetType() != cmStateEnums::INTERFACE_LIBRARY' failed.
Aborted (core dumped)

```

djaeckle, 2019-10-23:
Hi hovercraft-github,

i don't have an issue when i import the project inside KDevelop according to the wiki. Could you provide more information?

hovercraft-github, 2019-10-23:
Hi djaeckle!
Thank you for the reply.
Please take a note: I'm using ARM compiler and tools downloaded from [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads), and I put it here on my computer: "/home/al/local/gcc-arm-none-eabi-8-2019-q3-update".
My kdevelop is Version 5.2.1.
I will be grateful if you can tell me what exact versions of op.system, cmake, kdevelop and ARM tools are you using.
Also my question is: does ARM toolchain locations other than the system one supported? I.e. can I use my compiler from /home/al/local/gcc-arm-none-eabi-8-2019-q3-update anyhow?

djaeckle, 2019-10-23:
Hi,

my KDevelop is also 5.2.1. I have just verified it with version gcc-arm-none-eabi-8-2019-q3-update and its working properly. What you have to do is to update the TOOLCHAIN_PREFIX in your configuration.

In the wiki, we have:

> Follow the indications to setup the project and add -DCMAKE_TOOLCHAIN_FILE=cmake/toolchain-arm-none-eabi.cmake to the Extra Arguments.

You can append the extra arguments separated with spaces to:

`-DCMAKE_TOOLCHAIN_FILE=cmake/toolchain-arm-none-eabi.cmake -DTOOLCHAIN_PREFIX=/home/al/local/gcc-arm-none-eabi-8-2019-q3-update/`

hovercraft-github, 2019-10-23:
Yes, I did it, and project is created successfully, but when I'm trying to build, I'm getting the following:
```
/home/al/src/lora/LoRaMac-node/build> /usr/bin/cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_BUILD_TYPE=Debug -DCMAKE_TOOLCHAIN_FILE=cmake/toolchain-arm-none-eabi.cmake -DTOOLCHAIN_PREFIX=/home/al/local/gcc-arm-none-eabi-8-2019-q3-update/ /home/al/src/lora/LoRaMac-node
CMake Error at src/CMakeLists.txt:67 (include):
  include could not find load file:

    stm32l0

CMake Error at src/apps/LoRaMac/CMakeLists.txt:166 (include):
  include could not find load file:

    gdb-helper

CMake Error at src/apps/LoRaMac/CMakeLists.txt:167 (include):
  include could not find load file:

    binutils-arm-none-eabi

CMake Error at src/apps/LoRaMac/CMakeLists.txt:170 (generate_run_gdb_stlink):
  Unknown CMake command "generate_run_gdb_stlink".

-- Configuring incomplete, errors occurred!
See also "/home/al/src/lora/LoRaMac-node/build/CMakeFiles/CMakeOutput.log".
*** Failure: Exit code 1 ***
```

djaeckle, 2019-10-23:
Hi,

> /home/al/src/lora/LoRaMac-node/build> /usr/bin/cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_BUILD_TYPE=Debug [...]

looks valid. Have you removed the build folder before applying the new config?

mluis1, 2019-10-23:
You forgot `..` at the end of your command.

Your command should be:
```bash
/usr/bin/cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_BUILD_TYPE=Debug -DCMAKE_TOOLCHAIN_FILE=cmake/toolchain-arm-none-eabi.cmake -DTOOLCHAIN_PREFIX=/home/al/local/gcc-arm-none-eabi-8-2019-q3-update/ /home/al/src/lora/LoRaMac-node ..
```

Another point you are missing several required options in order to successfully build the project.

A valid command should look like the following (works for NucleoL152+SX1276MB1MAS platform):
```bash
cmake -DCMAKE_BUILD_TYPE=Debug \
           -DCMAKE_TOOLCHAIN_PREFIX=/home/al/local/gcc-arm-none-eabi-8-2019-q3-update
           -DCMAKE_TOOLCHAIN_FILE="../cmake/toolchain-arm-none-eabi.cmake" \
           -DAPPLICATION="LoRaMac" \
           -DCLASS="classA" \
           -DCLASSB_ENABLED="OFF" \
           -DACTIVE_REGION="LORAMAC_REGION_EU868" \
           -DREGION_EU868="ON" \
           -DREGION_US915="OFF" \
           -DREGION_CN779="OFF" \
           -DREGION_EU433="OFF" \
           -DREGION_AU915="OFF" \
           -DREGION_AS923="OFF" \
           -DREGION_CN470="OFF" \
           -DREGION_KR920="OFF" \
           -DREGION_IN865="OFF" \
           -DREGION_RU864="OFF" \
           -DBOARD="NucleoL152" \
           -DMBED_RADIO_SHIELD="SX1276MB1MAS" .. && make -j4
```

hovercraft-github, 2019-10-23:
@mluis1 This command is not my creation ) It is Kdevelops output log, and it contains the absolute path at the end instead of ... by the way, so it's valid.

hovercraft-github, 2019-10-23:
@djaeckle Yes, surely. Furthermore, if i run this command from the console, it works fine and then I can open the project in Kdevelop and build it successfully.
My mistake originally was to add the option -GKDevelop3 - it has been causing the crash.
But running cmake from inside of Kdevelop doesn't work for unknown reason.

hovercraft-github, 2019-10-23:
[image]
I'm tried to reproduce it on the different computer: Fedora 29, KDevelop Version 5.4.2, cmake version 3.14.5. The message above is shown within a second after I am pressing Ok button in the "Configure a build directory.." wizard.

djaeckle, 2019-10-25:
Hi,

> Yes, I did it, and project is created successfully, but when I'm trying to build, I'm getting the following:
> 
> ```
> /home/al/src/lora/LoRaMac-node/build> /usr/bin/cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_BUILD_TYPE=Debug -DCMAKE_TOOLCHAIN_FILE=cmake/toolchain-arm-none-eabi.cmake -DTOOLCHAIN_PREFIX=/home/al/local/gcc-arm-none-eabi-8-2019-q3-update/ /home/al/src/lora/LoRaMac-node
> CMake Error at src/CMakeLists.txt:67 (include):
>   include could not find load file:
> 
>     stm32l0
> 
> 
> CMake Error at src/apps/LoRaMac/CMakeLists.txt:166 (include):
>   include could not find load file:
> 
>     gdb-helper
> 
> 
> CMake Error at src/apps/LoRaMac/CMakeLists.txt:167 (include):
>   include could not find load file:
> 
>     binutils-arm-none-eabi
> 
> 
> CMake Error at src/apps/LoRaMac/CMakeLists.txt:170 (generate_run_gdb_stlink):
>   Unknown CMake command "generate_run_gdb_stlink".
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/al/src/lora/LoRaMac-node/build/CMakeFiles/CMakeOutput.log".
> *** Failure: Exit code 1 ***
> ```
> 
> @djaeckle Can you reproduce this?

i double checked it again and indeed i get the same issue as above directly after import. However, when i remove the build folder as a first action and perform the build afterwards, everything works.

Unfortunately, i cannot verify it on a Fedora system.

mluis1, 2020-02-21:
Can we close this issue?

mluis1, 2020-03-18:
No feedback, we close the issue.

closed 2020-03-18 by mluis1