Lora-net/LoRaMac-node #802: #define CRITICAL_SECTION_BEGIN( ): a label can only be part of a statement and a declaration is not a statement

VitorLima, 2019-10-15:
I am having this error:
utilities.h:138:35: error: a label can only be part of a statement and a declaration is not a statement
 #define CRITICAL_SECTION_BEGIN( ) uint32_t mask;  BoardCriticalSectionBegin( &mask )
The main problem is that I have already used an older version of LoRaMac-node, with the same declaration, and no error. 
I even build this version of lib with no error a code starting. 
I am suspicious it might be a wrong bracket but could no find it yet.
Any suggestion?

trick2011, 2019-10-16:
Where are you using the define as that seems to be the problem.

VitorLima, 2019-10-16:
Thank you @trick2011, you were right.
Actually the define was used in a case label. So it was like:

```
case DEVICE_RESTART:
CRITICAL_SECTION_BEGIN( );
do stuff
CRITICAL_SECTION_END( );
break;
```

Which could be resolved with the addition of brackets. I only realized that after put a "printf" before it for debugging and see the error gone.

closed 2019-10-16 by VitorLima