Lora-net/LoRaMac-node #803: crc not working with config sf6

mttbx, 2019-10-16:
I can receive wrong msg (unintelligible code) in radio with config sf6.
I set crc on and use implicit header (in both receive and send side).
So it seems payload crc may not working in sf6, am I right?

The Chip I use is sx1278.

mttbx, 2019-10-16:
Some pseudo codes are shown below. Just like pingpong code. The msg and response is carry with a counter, so I can know if the msg is correct.

tx process:
sx1278 init
loop:
  send_msg
  set_rx
  while i++ < 400:
      check if get response
      if get:
         printf the msg
      else:
         wait for 1ms
  wait for another 1s

rx process:
sx1278 init
set_rx
loop:
  check if get response
  if get:
     send_response
     reset_rx
     wait for 1ms
  else:
     wait for 1ms

power: 17, bandwith: 0(125000Hz), datarate: 6/7, cr: 2/4, fixlen: 1, payload: 40/50, crcon: 1, freqhopon: 0, iqinverted: 0, symbtimeout: 64

mluis1, 2019-10-25:
This question is more related to the radio usage and not to the LoRaWAN MAC layer implementation.

Could you please ask these kind of questions on [LoRa Developer Portal](https://lora-developers.semtech.com/knowledge-base/forum/)

closed 2019-10-25 by mluis1