Lora-net/LoRaMac-node #806: Tx Timeout

Devd2101, 2019-10-23:
Hi
I am doing porting of Lora Mac  Node  on STM32G070.
I have completed all the mentioned steps. the code is running  there is no error right Now.
But I always getting Tx TimeOut.

SSTPL
###### ===== ClassC demo application v1.0.RC1 ==== ######

Semtech Id 0x22

###### ===== JOINED ==== ######

ABP

Payload is :
40f93008010266759af4e1
Timeout 4000

###### ===== MCPS-Request ==== ######
STATUS      : OK

###### ===== MCPS-Confirm ==== ######
STATUS      : Tx timeout

###### ===== UPLINK FRAME 1 ==== ######

CLASS       : C

TX PORT     : 2
TX DATA     : UNCONFIRMED
05

DATA RATE   : DR_0
U/L FREQ    : 865985000
TX POWER    : 0
CHANNEL MASK: 0007

esantosjr, 2019-10-24:
Related issue: #685

The 'Tx timeout' message indicates that there is no connection between micro-controller and radio. One option is to verify if the physical connections match your pinout definitions (SPI number, MOSI, MISO, etc).

Devd2101, 2019-10-24:
Hi,

I am able to read the sx1272 register.

I have connected sx1272 on SPI1.

mluis1, 2019-10-25:
The SPI connections may be correctly done but, have you checked that the DIO pins are correctly connected and that the right interrupt handler is called when you observe a rising edge on the DIO pins?

Could you please check older issues which report the same kind of issues that you are observing?
Maybe you will find a solution to your problem by carefully read those older issues.

Sorry, but we can't really help you more on this.

closed 2019-10-25 by mluis1

Devd2101, 2019-10-30:
Thanks for suggestion.
I

robsoncouto, 2019-11-08:
Don't know if it helps you, but in case anyone finds this. 

I had a similar problem with STM32G071 + sx1262. The DIO1 pin was not triggering interrupts.
In my custom gpio-board.c file I had to implement

`HAL_GPIO_EXTI_Rising_Callback()`

Instead of

`HAL_GPIO_EXTI_Callback()`

The HAL_GPIO_EXTI_Callback signature does not seem to be present in stm32g0xx_hal_gpio.h

```
...
/* IO operation functions *****************************************************/
GPIO_PinState     HAL_GPIO_ReadPin(GPIO_TypeDef *GPIOx, uint16_t GPIO_Pin);
void              HAL_GPIO_WritePin(GPIO_TypeDef *GPIOx, uint16_t GPIO_Pin, GPIO_PinState PinState);
void              HAL_GPIO_TogglePin(GPIO_TypeDef *GPIOx, uint16_t GPIO_Pin);
HAL_StatusTypeDef HAL_GPIO_LockPin(GPIO_TypeDef *GPIOx, uint16_t GPIO_Pin);
void              HAL_GPIO_EXTI_IRQHandler(uint16_t GPIO_Pin);
void              HAL_GPIO_EXTI_Rising_Callback(uint16_t GPIO_Pin);
void              HAL_GPIO_EXTI_Falling_Callback(uint16_t GPIO_Pin);
...
```

Devd2101, 2019-11-08:
It's correct.

I also find similar and implemented same.

pho3nix06, 2023-07-26:
I know it's been almost 4 years, but has the issue been solved?

Devilal-Prajapat, 2023-07-26:
Resolved

Regards,

Devilal Prajapat

yaswanthksas, 2025-04-11:
[image]
hi sir, i am getting the issue of timeout please solve it for me as well in abp mode of stm32