Lora-net/LoRaMac-node #809: downlink musticast frame decrypted fail

wwx0715, 2019-10-29:
I test multicast  using branch feature/5.0.0(NucleoL476). Multicast setup is OK, but decryption is fail. I debug this problem and found fail at FOptsEncrypt (called at LoRaMacCryptoUnsecureMessage). For multicast frame, macMsg->FHDR.FCtrl.Bits.FOptsLen is 0, and fCntID is MC_FCNT_DOWN_0, but FOptsEncrypt  not consider this condition and just return LORAMAC_CRYPTO_FAIL_PARAM. I add a check at FOptsEncrypt  that return SUCCESS if size is 0. Please confirm this, tks.

mluis1, 2019-10-29:
At first sight it looks that you are right.
We need some time to further investigate this.

mluis1, 2020-03-09:
Could you please try the below patch?

I didn't had time to perform extensive tests on this yet. But, I think that it should fix the observed issue.

```patch
diff --git a/src/mac/LoRaMacCrypto.c b/src/mac/LoRaMacCrypto.c
index cfc09670..ffc5f091 100644
--- a/src/mac/LoRaMacCrypto.c
+++ b/src/mac/LoRaMacCrypto.c
@@ -1593,11 +1593,14 @@ LoRaMacCryptoStatus_t LoRaMacCryptoUnsecureMessage( AddressIdentifier_t addrID,
 #if( USE_LRWAN_1_1_X_CRYPTO == 1 )
     if( CryptoCtx.NvmCtx->LrWanVersion.Fields.Minor == 1 )
     {
-        // Decrypt FOpts
-        retval = FOptsEncrypt( macMsg->FHDR.FCtrl.Bits.FOptsLen, address, DOWNLINK, fCntID, fCntDown, macMsg->FHDR.FOpts );
-        if( retval != LORAMAC_CRYPTO_SUCCESS )
+        if( addrID == UNICAST_DEV_ADDR )
         {
-            return retval;
+            // Decrypt FOpts
+            retval = FOptsEncrypt( macMsg->FHDR.FCtrl.Bits.FOptsLen, address, DOWNLINK, fCntID, fCntDown, macMsg->FHDR.FOpts );
+            if( retval != LORAMAC_CRYPTO_SUCCESS )
+            {
+                return retval;
+            }
         }
     }
```

closed 2020-03-10 by mluis1