Lora-net/LoRaMac-node #811: Reset/Reinitialize LoRaMAC

wnts, 2019-10-31:
Is it possible to bring the LoRaMAC layer back into a known, default reset state, when it is 
in an arbitrary state?
If so, what API calls would have to be made to achieve this, and can this be considered a stable method in future releases?

darkfader, 2019-11-17:
I want to try this too (on some nodes that don't want to send anymore).
To be sure, I guess one solution would be to clear out the bss sections for the LoRa-stack only and call initialisation again. (and then probably do a (re-)join request again)
But perhaps the init functions already clear everything.

mluis1, 2020-03-18:
On commit a3726ab574ab72789b2450dcdd2cbe0b7ed3d14b we have added a new API named `LoRaMacDeInitialization`. 
The addition of this API should ease the MAC layer full reset operation.

To reset the MAC layer one could write something like the following:
```c
void MacLayerReset( void )
{
    // De-initialize the MAC layer
    LoRaMacStatus_t status = LORAMAC_STATUS_ERROR;
    do
    {
        status = LoRaMacDeInitialization( );
    }while( status != LORAMAC_STATUS_OK );

    // Re-initialize the MAC layer
    status = LoRaMacInitialization( &macPrimitives, &macCallbacks, ACTIVE_REGION );
    if ( status != LORAMAC_STATUS_OK )
    {
        printf( "LoRaMac wasn't properly initialized, error: %s", MacStatusStrings[status] );
        // Fatal error, endless loop.
        while ( 1 )
        {
        }
    }
}
```

closed 2020-03-18 by mluis1