Lora-net/LoRaMac-node #814: Too many missed frames recovery

ChrisEAlfred, 2019-11-01:
When the MAC detects too many missed frames (LORAMAC_EVENT_INFO_STATUS_DOWNLINK_TOO_MANY_FRAMES_LOSS), how does the stack recover.

This can happen if FCnt jumps (e.g. gateway restarted). There was a 3rd party network server update and FCnt was reset. Now we longer have uplinks.

Should the MAC automatically recover, or do we need to do something?

mluis1, 2019-11-01:
The only way to recover from this situation is to perform a new activation in order to get everything in sync again.

The MAC layer itself cannot automatically recover from this situation. The only thing it can do is to notify the upper layer about the encountered issue. In this case `LORAMAC_EVENT_INFO_STATUS_DOWNLINK_TOO_MANY_FRAMES_LOSS` the upper layer may decide to perform a new activation (issuing a JoinReq).

If OTAA is used as activation method the recovery is easy as the node just needs to join again the network.

If ABP is used this situation introduces security issues. As the network server FCnt values have been reset the same counter values will be re-used with the same set of keys (the security issue). This means that one must update the set of keys for each ABP device in order to be able to reset the FCnt counter.

But, the original issue is on the server side. The network operator must ensure that the end-devices contexts aren't lost in abnormal situations. (server updates, power loss, etc...).

Please also note that this is one of the reasons why OTAA should be preferred over ABP activation.

mluis1, 2020-02-21:
Can we close this issue?

mluis1, 2020-03-18:
No feedback, we close the issue.

closed 2020-03-18 by mluis1