Lora-net/LoRaMac-node #815: Class B RX window not opened until first transmit

jcwren, 2019-11-01:
I'm not clear if this is a usage error from my end, or an issue in the stack. I know part of the problem lies in the STMicro implementation of LoRaWAN on top of LoRaMac, but I'm hoping for some insight here.

When the application first comes up, it does a successful OTAA join. It then tries switch to class B with `LORAWAN_DEFAULT_PING_SLOT_PERIODICITY` = 4 (16 seconds). The beacon is acquired, as shown from this debug fragment below. However, after the beacon is locked, until the first time the application transmits, the downlink receive window that should be opening every 16 seconds does not. The first time the application sends data (typically on a 1-hour interval), then the RX window starts opening for true class B operation (2nd debug output). 

I would expect the RX window to start opening every 16 seconds immediately after the beacon lock has been attained, not require the application to send data to enable it (unless I'm misunderstanding something in the specification).

Unfortunately, in the stock STMicro LoRaWAN code, there's no callback to the application when the beacon is locked or lost, so without modifying it (which I generally try to avoid because of how they upgrade releases), the application doesn't that now that it's in class B that it should send a packet to start the RX window opening.

#### Device joins successfully, switches to class B, has beacon lock, but is not opening RX window every 16 seconds:
```
  0s726: mlmeRequest->Type=MLME_JOIN/0
  0s765: TX on freq 903700000 Hz at DR 0
  0s778: LoRaMacIsBusy(): MacCtx.MacState=2, MacCtx.AllowRequests=1
  0s779: LoRaMacMlmeRequest()/4496: returning LORAMAC_STATUS_BUSY
  1s149: PHY txDone
  6s130: RX on freq 927500000 Hz at DR 10 (maxPayload 242)
  6s264: PHY rxDone
  6s270: APP> MlmeConfirm STATUS: OK
  6s270: Joined, switching to class B
  6s271: mlmeRequest->Type=MLME_BEACON_ACQUISITION/11
  6s276: Starting class B acquisition
  6s279: mlmeRequest->Type=MLME_DEVICE_TIME/9
 20s048: GPS fix acquired in 19264ms (shortest was 19264ms, longest was 19264ms)
134s292: APP> MlmeConfirm STATUS: Beacon not found
134s293: LORA_BeaconReq ()
134s293: mlmeRequest->Type=MLME_BEACON_ACQUISITION/11
134s298: Starting class B acquisition
262s311: APP> MlmeConfirm STATUS: Beacon not found
262s312: LORA_BeaconReq ()
262s312: mlmeRequest->Type=MLME_BEACON_ACQUISITION/11
262s317: Starting class B acquisition
390s330: APP> MlmeConfirm STATUS: Beacon not found
390s331: LORA_BeaconReq ()
390s331: mlmeRequest->Type=MLME_BEACON_ACQUISITION/11
390s335: Starting class B acquisition
518s348: APP> MlmeConfirm STATUS: Beacon not found
518s349: LORA_BeaconReq ()
518s349: mlmeRequest->Type=MLME_BEACON_ACQUISITION/11
518s354: Starting class B acquisition
646s367: APP> MlmeConfirm STATUS: Beacon not found
646s368: LORA_BeaconReq ()
646s368: mlmeRequest->Type=MLME_BEACON_ACQUISITION/11
646s373: Starting class B acquisition
774s385: APP> MlmeConfirm STATUS: Beacon not found
774s386: LORA_BeaconReq ()
774s386: mlmeRequest->Type=MLME_BEACON_ACQUISITION/11
774s391: Starting class B acquisition
843s950: PHY rxDone
843s951: APP> MlmeConfirm STATUS: OK
843s952: LORA_PingSlotReq ()
843s954: mlmeRequest->Type=MLME_LINK_CHECK/3
843s960: mlmeRequest->Type=MLME_PING_SLOT_INFO/12
843s968: APP> MLMEInd STATUS: Beacon locked
843s972: #= BEACON 1256589312 =#, GW desc 0, rssi -75, snr 1, freq 923300000
971s963: PHY rxDone
971s965: APP> MLMEInd STATUS: Beacon locked
971s966: #= BEACON 1256589440 =#, GW desc 0, rssi -76, snr 1, freq 923900000
1099s978: PHY rxDone
1099s980: APP> MLMEInd STATUS: Beacon locked
1099s980: #= BEACON 1256589568 =#, GW desc 0, rssi -71, snr 2, freq 924500000
1227s993: PHY rxDone
```
#### Beacon still locked, application sends data, stack starts opening RX window every 16 seconds:
```
3404s238: APP> MLMEInd STATUS: Beacon locked
3404s239: #= BEACON 1256591872 =#, GW desc 0, rssi -74, snr 1, freq 925700000
3532s250: PHY rxDone
3532s251: Assigned != 1
3532s251: APP> MLMEInd STATUS: Beacon locked
3532s252: #= BEACON 1256592000 =#, GW desc 0, rssi -75, snr 1, freq 926300000
3627s365: GPS fix acquired in 16587ms (shortest was 16587ms, longest was 19264ms)
3627s367: Sending 0x80 0x02 0x09 0xb7 0xa4 0xfa 0xff 0x27 0xc9 (9 bytes) to host port 3
3627s374: LORA_send() is sending empty frame to flush MAC commands
3627s383: TX on freq 902500000 Hz at DR 0
3627s725: PHY txDone
3628s707: RX on freq 923900000 Hz at DR 10 (maxPayload 242)
3628s850: PHY rxDone
3628s854: SrvMacCmd=SRV_MAC_DEVICE_TIME_ANS/13
3628s855: We got beacon!
3628s855: SrvMacCmd=SRV_MAC_LINK_CHECK_ANS/2
3628s859: SrvMacCmd=SRV_MAC_PING_SLOT_INFO_ANS/16
3628s863: SrvMacCmd=SRV_MAC_LINK_ADR_REQ/3
3628s867: SrvMacCmd=SRV_MAC_PING_SLOT_CHANNEL_REQ/17
3628s872: APP> McpsConfirm STATUS: OK
3628s875: #= U/L FRAME 1 =# Class A, Port 3, data size 9, pwr 5, Channel Mask 00FF 0000 0000 0000 0000
3628s883: APP> MlmeConfirm STATUS: OK
3628s887: APP> MlmeConfirm STATUS: OK
3628s890: #= Switch to Class B done. =#
3628s894: Switch to class B done
3628s897: APP> MlmeConfirm STATUS: OK
3628s900: APP> McpsInd STATUS: OK
3628s903: #= D/L FRAME 0 =# RxWin 1, Port 0, data size 0, rssi -60, snr 9
3628s910: Sending 0xa0 (1 bytes) to host port 1
3628s916: TX on freq 902700000 Hz at DR 0
3629s300: PHY txDone
3630s282: RX on freq 924500000 Hz at DR 10 (maxPayload 242)
3630s395: PHY rxDone
3630s397: SrvMacCmd=SRV_MAC_LINK_ADR_REQ/3
3630s398: SrvMacCmd=SRV_MAC_PING_SLOT_CHANNEL_REQ/17
3630s400: APP> McpsConfirm STATUS: OK
3630s403: #= U/L FRAME 2 =# Class B, Port 1, data size 1, pwr 5, Channel Mask 00FF 0000 0000 0000 0000
3630s412: APP> McpsInd STATUS: OK
3630s415: #= D/L FRAME 1 =# RxWin 1, Port 0, data size 0, rssi -64, snr 8
3632s210: RX on freq 925100000 Hz at DR 8 (maxPayload 53)
3632s291: PHY rxTimeOut
3647s571: RX on freq 925100000 Hz at DR 8 (maxPayload 53)
3647s651: PHY rxTimeOut
3660s264: PHY rxDone
3660s266: APP> MLMEInd STATUS: Beacon locked
3660s266: #= BEACON 1256592128 =#, GW desc 0, rssi -75, snr 1, freq 926900000
3676s162: RX on freq 925700000 Hz at DR 8 (maxPayload 53)
3676s242: PHY rxTimeOut
3691s521: RX on freq 925700000 Hz at DR 8 (maxPayload 53)
3691s601: PHY rxTimeOut
3706s881: RX on freq 925700000 Hz at DR 8 (maxPayload 53)
3706s961: PHY rxTimeOut
3722s242: RX on freq 925700000 Hz at DR 8 (maxPayload 53)
```

reissjason, 2019-11-01:
The uplink is needed with ClassB bit set in the LoRaWAN Header FCtrl field to notify the network that the device is ready to accept downlinks via Class B windows.
Also a PingSlotInfoReq is needed to inform the network of the window periodicity.

There appears to be some activity in this regard minus the uplink.

> 843s951: APP> MlmeConfirm STATUS: OK
> 843s952: LORA_PingSlotReq ()
> 843s954: mlmeRequest->Type=MLME_LINK_CHECK/3
> 843s960: mlmeRequest->Type=MLME_PING_SLOT_INFO/12

jcwren, 2019-11-01:
OK. I've seen that get stuffed in the buffer, and wondered why it didn't transmit. In the class B examples, I see the state machine goes straight from the beacon acquisition to sending data. But shouldn't the MAC layer whichever flags necessary so that the request to transmit is propagated up the stack, even if the MAC commands are intended to be piggybacked? Otherwise the application doesn't know there's any pending to transmit.

I was trying to follow the flags and various calls to figure out if there was somewhere I can pass the notification up, but all that is a bit convoluted and I keep losing track of where I am in the various states...

mluis1, 2020-02-21:
Can we close this issue?

jcwren, 2020-02-21:
I'll have to diff the changes since I posted this against the current stack. I haven't been following what changes have been posted, so don't know if there's been a callback added to notify the application that stack is now in a class B state.

Basically, the short form:
1) Application joins network
2) Stack hears class B beacon, puts data in buffer to be transmitted
3) Stack fails to notify application it's ready to operate in class B mode
4) Actual class B operation doesn't start until next time the application transmits, which could be days.

mluis1, 2020-02-24:
When an end-device switch to ClassB the application layer gets notified. This functionality is provided by the example codes since the ClassB implementation has been provided the first time on this project.

Please take a look at the below code excerpts on the provided examples. As you can observe once the notification is received an uplink can be scheduled as soon as possible to notify the network server about the new end-device state.

Please also note that the ST en.i-cube_lrwan package has similar functionality in its examples.

classB example [main.c](https://github.com/Lora-net/LoRaMac-node/blob/develop/src/apps/LoRaMac/classB/NucleoL073/main):
```c
static void MlmeConfirm( MlmeConfirm_t *mlmeConfirm )
{
...
        case MLME_PING_SLOT_INFO:
        {
            if( mlmeConfirm->Status == LORAMAC_EVENT_INFO_STATUS_OK )
            {
                mibReq.Type = MIB_DEVICE_CLASS;
                mibReq.Param.Class = CLASS_B;
                LoRaMacMibSetRequestConfirm( &mibReq );

                printf( "\r\n\r\n###### ===== Switch to Class B done. ==== ######\r\n\r\n" );

                WakeUpState = DEVICE_STATE_SEND;
                DeviceState = WakeUpState;
                NextTx = true;
            }
            else
            {
                WakeUpState = DEVICE_STATE_REQ_PINGSLOT_ACK;
            }
            break;
...
}
```

periodic-uplink-lpp example [main.c](https://github.com/Lora-net/LoRaMac-node/blob/develop/src/apps/LoRaMac/periodic-uplink-lpp/NucleoL073/main.c)
```c
static void OnClassChange( DeviceClass_t deviceClass )
{
    DisplayClassUpdate( deviceClass );

    // Inform the server as soon as possible that the end-device has switched to ClassB
    LmHandlerAppData_t appData =
    {
        .Buffer = NULL,
        .BufferSize = 0,
        .Port = 0
    };
    LmHandlerSend( &appData, LORAMAC_HANDLER_UNCONFIRMED_MSG );
}
```

jcwren, 2020-02-24:
It doesn't work like you seem to think it does. The application requests the change to class B. The stack will look for the beacon, and then place several messages (`SRV_MAC_LINK_CHECK_ANS`, `SRV_MAC_PING_SLOT_INFO_ANS`, `SRV_MAC_LINK_ADR_REQ`, `SRV_MAC_PING_SLOT_CHANNEL_REQ`) in the buffer to be sent. The stack does NOT inform the application that it has locked to the beacon and there are now pending bytes in the buffer to be sent. 

Once the application DOES send a packet, those pending bytes are sent, and the full switch to class B completes. THEN the stack notifies the application as you indicate. I've been well aware of that for some time. Their examples all assume that the application will attempt transmit a message of their own BEFORE the full class B negotiation completes, which then causes it to ACTUALLY complete.

The solution that I've found is to modify `MlmeIndication()` so the `MLME_BEACON` case calls `LoRaMainCallbacks->LORA_TxNeeded()`. I've modified the `LORA_TxNeeded` callback to pass a parameter indicating if it's a beacon lock, scheduled uplink, or a frame flush. In my application I check if the current class is the same as the desired class or not if it is a beacon lock. If it's not, then I force a `LORA_send()` call to flush the buffer. This works well.

Waiting for the application to decide it's ready to transmit something to flush the buffer presents two problems. The switch to mode B can take up to 24 hours (my application transmits as little as once a day), which means that during the 24 hour window it is not possible to send a command to the device and have it act on it, and you're limited to 11 byte data buffers if using DR0, If the class B MAC messages are pending, it's not always possible to put a complete application message in the buffer, which requires managing a retry.

mluis1, 2020-04-22:
The MAC layer does inform the upper layer if the beacon is locked or not.

Extracted from class B example:
```c
        case MLME_BEACON_ACQUISITION:
        {
            if( mlmeConfirm->Status == LORAMAC_EVENT_INFO_STATUS_OK )
            {
                WakeUpState = DEVICE_STATE_REQ_PINGSLOT_ACK;
            }
            else
            {
#if defined( USE_BEACON_TIMING )
                WakeUpState = DEVICE_STATE_REQ_BEACON_TIMING;
#else
                WakeUpState = DEVICE_STATE_REQ_DEVICE_TIME;
#endif
            }
            break;
        }
```
After setting the `WakeupState` to `DEVICE_STATE_REQ_PINGSLOT_ACK` the application will execute
```c
            case DEVICE_STATE_REQ_PINGSLOT_ACK:
            {
                MlmeReq_t mlmeReq;

                if( NextTx == true )
                {
                    mlmeReq.Type = MLME_LINK_CHECK;
                    LoRaMacMlmeRequest( &mlmeReq );

                    mlmeReq.Type = MLME_PING_SLOT_INFO;
                    mlmeReq.Req.PingSlotInfo.PingSlot.Fields.Periodicity = LORAWAN_DEFAULT_PING_SLOT_PERIODICITY;
                    mlmeReq.Req.PingSlotInfo.PingSlot.Fields.RFU = 0;

                    if( LoRaMacMlmeRequest( &mlmeReq ) == LORAMAC_STATUS_OK )
                    {
                        WakeUpState = DEVICE_STATE_SEND;
                    }
                }
                DeviceState = DEVICE_STATE_SEND;
                break;
            }
```
Here the application just request to send the MAC command on the next uplink. Then the application state is set to `DEVICE_STATE_SEND` where the uplink effectively is sent.

Once the `MLME_PING_SLOT_INFO` answer is received the application layer will ask the MAC layer to switch to Class B.
```c
        case MLME_PING_SLOT_INFO:
        {
            if( mlmeConfirm->Status == LORAMAC_EVENT_INFO_STATUS_OK )
            {
                mibReq.Type = MIB_DEVICE_CLASS;
                mibReq.Param.Class = CLASS_B;
                LoRaMacMibSetRequestConfirm( &mibReq );

                printf( "\n\n###### ===== Switch to Class B done. ==== ######\n\n" );

                WakeUpState = DEVICE_STATE_SEND;
                DeviceState = WakeUpState;
                NextTx = true;
            }
            else
            {
                WakeUpState = DEVICE_STATE_REQ_PINGSLOT_ACK;
            }
            break;
        }
```

At the next uplink the MAC layer will inform the network server that it is in ClassB mode by setting the ClassB bit in the uplink frame FCtrl field.
Once this uplink has been sent the MAC layer starts opening the ping-slot reception windows.

mluis1, 2020-11-26:
No feedback we close this issue.

closed 2020-11-26 by mluis1