Lora-net/LoRaMac-node #816: ack missing in uplink after downlink with ack request

steph44fr, 2019-11-05:
ack missing in uplink after downlink with ack request

after receiving downlink with ack request I send uplink for ack
in SendFrame( void ) call of LoRaMacMcpsRequest return LORA_MAC_STATUS_DUTYCYCLE_RESTRICTED due to Duty cycle

so periodicaly I call SendFrame() until uplink is send, but this uplink doesnit contain the ack

steph44fr, 2019-11-05:
stack is version 4.4.1

steph44fr, 2019-11-05:
// SEND, IsTxConfirmed = TRUE
```
static b SendFrame( void )
{
	McpsReq_t mcpsReq;
	LoRaMacTxInfo_t txInfo;
	if( LoRaMacQueryTxPossible( AppDataSize, &txInfo ) != LORAMAC_STATUS_OK ) {
		// Send empty frame in order to flush MAC commands
		mcpsReq.Type = MCPS_UNCONFIRMED;
		mcpsReq.Req.Unconfirmed.fBuffer = NULL;
		mcpsReq.Req.Unconfirmed.fBufferSize = 0;
		mcpsReq.Req.Unconfirmed.Datarate = lora_parameters.default_DR;
	} else {
		if( IsTxConfirmed == false) {
			mcpsReq.Type = MCPS_UNCONFIRMED;
			mcpsReq.Req.Unconfirmed.fPort = AppPort;
			mcpsReq.Req.Unconfirmed.fBuffer = AppData;
			mcpsReq.Req.Unconfirmed.fBufferSize = AppDataSize;
			mcpsReq.Req.Unconfirmed.Datarate = LORAWAN_DEFAULT_DATARATE;
		} else {
			mcpsReq.Type = MCPS_CONFIRMED;
			mcpsReq.Req.Confirmed.fPort = AppPort;
			mcpsReq.Req.Confirmed.fBuffer = AppData;
			mcpsReq.Req.Confirmed.fBufferSize = AppDataSize;
			mcpsReq.Req.Confirmed.NbTrials = 8;
			mcpsReq.Req.Confirmed.Datarate = LORAWAN_DEFAULT_DATARATE;
		}
	}
	LoRaMacStatus_t status = LoRaMacMcpsRequest( &mcpsReq );
	if(status == LORAMAC_STATUS_OK )
	{
		return false;
	}
	return TRUE;
}
```

// TASK LoRa
```
case DEVICE_STATE_SEND:
	if( NextTx == true )
	{
		NextTx = SendFrame( );
		if(NextTx == TRUE)
		{
			if( xTimerStart( Send_Wait_DC, 0 ) != pdPASS )
			{
			}
			else
			{
			}
		}
	}
break;
```

// TIMER CALLBACK'void vTimerSendWaitDCCallback(void)
```
{
	xTaskNotify(lora_task_handle, DEVICE_STATE_SEND, eSetValueWithOverwrite);
	xTimerStop( Send_Wait_DC, 0 );
}
```

steph44fr, 2019-11-05:
before calling SendFrame for the first time, I call LoRaMacGetLoRaMacState wich return LORAMAC_IDLE,
the next call to SendFrame (vTimerSendWaitDCCallback) are without call LoRaMacGetLoRaMacState

steph44fr, 2019-11-13:
no need to implement periodicaly call to SendFrame()

in LoRaMacStatus_t Send( LoRaMacHeader_t *macHdr, uint8_t fPort, void *fBuffer, uint16_t fBufferSize )
replace 
status = ScheduleTx( false);
with
status = ScheduleTx( true );

it's seems OK

djaeckle, 2020-03-06:
Hi steph44fr,

thanks for the report. We have verified the operation with integration tests and we cannot reproduce the issue. In addition, the reset of the acknowledgement status in the code will be only reset, if the transmission was successful. 

> status = ScheduleTx( true );

I recommend to not change this for a fix, as this has a different purpose.

closed 2020-03-26 by mluis1