Lora-net/LoRaMac-node #817: SX1261 WakeUp problem

denisn73, 2019-11-07:
Found a problem with waking up the SX1261 chip.
In the sx1261mbxbas-board.c file of the void SX126xWakeup (void) function, you must add a delay of at least 100 μs after the ChipSelect signal.
`void SX126xWakeup( void )
{
    CRITICAL_SECTION_BEGIN( );

    GpioWrite( &SX126x.Spi.Nss, 0 );

    **delay_ms(1);** // <-- add this!!!

    SpiInOut( &SX126x.Spi, RADIO_GET_STATUS );
    SpiInOut( &SX126x.Spi, 0x00 );

    GpioWrite( &SX126x.Spi.Nss, 1 );

    // Wait for chip to be ready.
    SX126xWaitOnBusy( );

    CRITICAL_SECTION_END( );
}`
There is no such problem with the cx 262 chip, but in the datasheet this delay is indicated for both chips and should be added to the code.

mluis1, 2020-03-04:
Can we close this issue?

I was told that you received support on this issue from Semtech support team.

The current implementation is correct according to my colleagues.

mluis1, 2020-03-18:
No feedback. We close the issue.

closed 2020-03-18 by mluis1