Lora-net/LoRaMac-node #821: [example applications] Should LoRaMac classX check the returned value of LoRaMacInitialization?

C47D, 2019-11-11:
Hi,

Last week I was able to setup the tools for development on Windows 10. After some time of trying to get the LoRaMac-classA application running on my NucleoL476 board I found out I wasn't configuring it properly, depricted in [this](https://github.com/Lora-net/LoRaMac-node/issues/679#issuecomment-552008901) comment.

As a new user of the library I think printing a message in case `LoRaMacInitialization` not returning `LORAMAC_STATUS_OK` it would be useful. Something like below:

```c
LoRaMacStatus_t
 init_status = LoRaMacInitialization( &macPrimitives, &macCallbacks, ACTIVE_REGION );

if (LORAMAC_STATUS_OK != init_status) {
    printf("LoRaMac wasn't properly initialized, error: %s", MacStatusStrings[init_status]);
    assert(0);
}
```

Is it a good idea? I can send a PR in case you think it's helpful to new users like me.

mluis1, 2019-11-11:
Looks to be a good improvement.

Please propose a PR and we will review it.
Please also don't forget to update all provided examples in your PR.

Just one comment please follow the coding rules in place as much as possible.

The provide example would look like as shown below. 

- The status variable is already declared, no need for a new declaration.
- The assert function isn't declared by default. Would replace it by `while( 1 ){ }`

```c
status = LoRaMacInitialization( &macPrimitives, &macCallbacks, ACTIVE_REGION );

if( status != LORAMAC_STATUS_OK )
{
    printf( "LoRaMac wasn't properly initialized, error: %s", MacStatusStrings[status ] );
    // Fatal error. Endless loop.
    while( 1 )
    {
    }
}
```

C47D, 2019-11-11:
Cool, I will send a PR with your suggestions. Last question, against what branch should I do the PR?

mluis1, 2019-11-12:
Please do it against develop branch.

closed 2019-11-12 by mluis1