Lora-net/LoRaMac-node #824: Notified when transmission has been done in unconfirmed message mode

aydogane, 2019-11-13:
How can I be informed when the transmission has been done in unconfirmed message mode without modifying the lora.c file?

I have the following solution to this issue but I am looking for a callback solution because I don't want to (and also I am not supposed to) modify the stack.

Modifying the `McpsConfirm` function as below (adding a callback to the `LoRaMainCallback_t` type and calling it when the event info is OK for an unconfirmed message) works perfectly:
```
switch( mcpsConfirm->McpsRequest )
{
    case MCPS_UNCONFIRMED:
    {
        // Check Datarate
        // Check TxPower
        if (LoRaMainCallbacks->LORA_TransmitOk != NULL) {
            LoRaMainCallbacks->LORA_TransmitOk();
        }
        break;
    }
    case MCPS_CONFIRMED:
    {
        // Check Datarate
        // Check TxPower
        // Check AckReceived
        // Check NbTrials
        break;
    }
    case MCPS_PROPRIETARY:
    {
        break;
    }
    default:
        break;
}
```
but I am looking for a cleaner solution like passing a user defined callback by `LORA_Init` function or something like that.

Is what I ask possible or in any case do I need to modify the stack for this?

Thanks in advance

mluis1, 2019-11-13:
I guess that you are not using the files provided on this project.
In this project there is no `lora.c` file and `LoRaMainCallbacks` isn't defined.

Please ask this question to the provider of your stack.

In this project all provided examples have something similar to what you would like to have implemented.
`classA/B/C` examples (`main.c`)
```c
/*!
 * \brief   MCPS-Confirm event function
 *
 * \param   [IN] mcpsConfirm - Pointer to the confirm structure,
 *               containing confirm attributes.
 */
static void McpsConfirm( McpsConfirm_t *mcpsConfirm )
{
    printf( "\r\n###### ===== MCPS-Confirm ==== ######\r\n" );
    printf( "STATUS      : %s\r\n", EventInfoStatusStrings[mcpsConfirm->Status] );
    if( mcpsConfirm->Status != LORAMAC_EVENT_INFO_STATUS_OK )
    {
    }
    else
    {
        switch( mcpsConfirm->McpsRequest )
        {
            case MCPS_UNCONFIRMED:
            {
                // Check Datarate
                // Check TxPower
                break;
            }
            case MCPS_CONFIRMED:
            {
                // Check Datarate
                // Check TxPower
                // Check AckReceived
                // Check NbTrials
                break;
            }
            case MCPS_PROPRIETARY:
            {
                break;
            }
            default:
                break;
        }

        // Switch LED 1 ON
        GpioWrite( &Led1, 1 );
        TimerStart( &Led1Timer );
    }
    MibRequestConfirm_t mibGet;
    MibRequestConfirm_t mibReq;

    mibReq.Type = MIB_DEVICE_CLASS;
    LoRaMacMibGetRequestConfirm( &mibReq );

    printf( "\r\n###### ===== UPLINK FRAME %lu ==== ######\r\n", mcpsConfirm->UpLinkCounter );
    printf( "\r\n" );

    printf( "CLASS       : %c\r\n", "ABC"[mibReq.Param.Class] );
    printf( "\r\n" );
    printf( "TX PORT     : %d\r\n", AppData.Port );

    if( AppData.BufferSize != 0 )
    {
        printf( "TX DATA     : " );
        if( AppData.MsgType == LORAMAC_HANDLER_CONFIRMED_MSG )
        {
            printf( "CONFIRMED - %s\r\n", ( mcpsConfirm->AckReceived != 0 ) ? "ACK" : "NACK" );
[... 163 lines cut ...]
                printf( "%04X ", mibGet.Param.ChannelsMask[0] );
                break;
            }
            case LORAMAC_REGION_AU915:
            case LORAMAC_REGION_CN470:
            case LORAMAC_REGION_US915:
            {
                for( uint8_t i = 0; i < 5; i++)
                {
                    printf( "%04X ", mibGet.Param.ChannelsMask[i] );
                }
                break;
            }
            default:
            {
                printf( "\r\n###### ========= Unknown Region ============ ######" );
                break;
            }
        }
        printf("\r\n");
    }

    printf( "\r\n" );
}
```

closed 2019-11-13 by mluis1

aydogane, 2019-11-14:
Thanks a lot for your detailed explanation.