Lora-net/LoRaMac-node #825: Does the LoRaMac code for STM32 support USB_CDC communication with laptop?

liuming15, 2019-11-19:
I found some code in the uart-board.c named UartUsbPutChar( obj, data ) seemed to transfer data via USB communication. But I can't find the definition of UartUsbPutChar( obj, data ). Is it support USB CDC communication? Any help will be appreciated.

uint8_t UartMcuPutChar( Uart_t *obj, uint8_t data )
{
    if( obj->UartId == UART_USB_CDC )
    {
#if defined( USE_USB_CDC )
        return UartUsbPutChar( obj, data );
#else
        return 255; // Not supported
#endif
    }
    else
    {
        CRITICAL_SECTION_BEGIN( );
        TxData = data;

        if( IsFifoFull( &obj->FifoTx ) == false )
        {
            FifoPush( &obj->FifoTx, TxData );

            // Trig UART Tx interrupt to start sending the FIFO contents.
            __HAL_UART_ENABLE_IT( &UartHandle, UART_IT_TC );

            CRITICAL_SECTION_END( );
            return 0; // OK
        }
        CRITICAL_SECTION_END( );
        return 1; // Busy
    }
}

mluis1, 2019-11-19:
In earlier versions of this project we had support for the USB handling but we removed it because the currently supported platforms don't require it anymore..

The last version to have USB support implementation is version [4.4.1](https://github.com/Lora-net/LoRaMac-node/releases/tag/v4.4.1)

You may look at LoRaMote or SensorNode platforms implementation.

liuming15, 2019-11-19:
Great, I will try. Thanks.

> In earlier versions of this project we had support for the USB handling but we removed it because the currently supported platforms don't require it anymore..
>
> The last version to have USB support [...]

closed 2019-11-20 by mluis1