Lora-net/LoRaMac-node #826: UART RX not working on B-L072Z-LRWAN1

josesimoes, 2019-11-20:
Hi,

The UART RX is not working only the 1st char is received.

Suspect this is because of the way the UART interrupt is being handled...
The repeated call to `HAL_UART_Receive_IT` inside `HAL_UART_RxCpltCallback` is doomed to fail because of the way the ST UART HAL code works. There are various entries about this on ST forums and Stackoverflow too.

Can this be reworked to implement a workaround for those shortcomings? Or maybe switching to DMA...

zyk6271, 2019-11-27:
I meet this questions yesterday , I have solved it , let me tell you . In uart-board.c with line 220 . The name of func is UartMcuGetChar . Modify it to what I show to you. Here are the code.
`
uint8_t UartMcuGetChar( Uart_t *obj, uint8_t *data )
{
			if( IsFifoEmpty( &obj->FifoRx ) == false )
			{
					*data = FifoPop( &obj->FifoRx );
					HAL_Delay(1);
					return 0;
			}
			else
			{
					return 1;	
			}
}
`

mluis1, 2020-02-21:
Can we close this issue ?

josesimoes, 2020-02-21:
@mluis1 I still can't receive on the UART...

kevenmalter, 2020-03-24:
Hi @josesimoes ,did you resoved the problem?

kevenmalter, 2020-03-24:
Hi @josesimoes did you resolved the problem?

josesimoes, 2020-03-24:
Hi @kevenmalter the suggestion above seems to work. Shouldn't it be ported to the respective code file?

mluis1, 2021-12-06:
If you don't mind we decided to move this issue to the [Discussion](https://github.com/Lora-net/LoRaMac-node/discussions) tab as we are not able to reproduce this issue using the provided examples.