Lora-net/LoRaMac-node #827: fuota-test-01 McGroupSetupReq Not happening (Need Help)..

sathishkumarparthiban, 2019-11-20:
Hi ,
I'm trying to get the fuota-test-01 example to work. Steps i followed.
1. Created a device on the loraserver (OTAA).
2. Waited for the device to Join via OTAA.
3. After joining created a firmware update job using "create frimware update job" button .

After this i dont see "McGroupSetupReq & Ans" happening. after sometime i get a error message saying "The device failed to provision the remote multicast setup". (attached image)

Project :  fuota-test-01
Board : B-L072Z-LRWAN1
Gateway : n-fuse 868Mhz (IN865-867 region).
Server: loraserver

I
[image]

EduardoPfeifer, 2019-11-20:
Following...

mluis1, 2019-11-27:
Last week I tried for the first time the recent implementation of the FUOTA packages by ChirpStack.
I wasn't able either to completely run the test up until completion.

In my case the Multicast session was setup correctly but, then the Fragmentation session setup failed.

1. The end-device answered that there was not enough memory available on `FragSessionSetupAns`.
2. I increased the acceptable memory size by changing `FRAG_MAX_NB` and `FRAG_MAX_SIZE`.
3. Re-compiled the code and tried again and this time the end-device reported success on `FragSessionSetupAns`
4. The end-device switched to Class C but, didn't receive the downlinks.
5. I tried to analyze if the parameters provided by the server were correct and I noticed some strange values were provided by `FragSessionSetupReq`.
Unfortunately, I didn't had the time to investigate further as I had to move to another project.
Once my current tasks are finished I will try to understand and debug what is going on.

Please, remark that for the embedded side I'm confident that there is no issue (but, we never know) as this firmware has been used by different network stack providers while performing interoperability tests. All the network stack providers being part of those tests reported that everything was OK for the tested scenario.

On the other side the FUOTA support on ChirpStack is pretty recent and I currently don't know what is the development status of this feature on the server.

Please note, that the first tests done on the packages provided by this repository were done using a previous version of ChirpStack and by following the instructions provided by this [mbed](https://github.com/ARMmbed/mbed-os-example-lorawan-fuota/tree/master/fuota-server) project. I had to change the `loraserver.js` file in order to make it work. 
Please refer, to issue #640 for changes done on `loraserver.js`.

sathishkumarparthiban, 2019-11-27:
Hi mluis,

Thanks for the reply and i had used your modification on the loraserver.js file to run it.
I found that the msgs for "202" are not been posted to mqtt topics. so the loraserver.js script couldnt start the mcgroupreq..
I have posted here with my finding but apparently the post wasnt relative there and i need to close .
[https://github.com/ARMmbed/mbed-os-example-lorawan-fuota/issues/10](https://github.com/ARMmbed/mbed-os-example-lorawan-fuota/issues/10)

Can you show me your output of the loraserver.js on a successful run??

mluis1, 2019-11-27:
Please refer to the [readme.md](https://github.com/Lora-net/LoRaMac-node/blob/develop/src/apps/LoRaMac/fuota-test-01/readme.md) file under fuota-test-01 directory on this project.

The [readme.md](https://github.com/Lora-net/LoRaMac-node/blob/develop/src/apps/LoRaMac/fuota-test-01/readme.md) file provides the full log of a successful session using the `loraserver.js` file.

KaDw, 2019-11-30:
@mluis1 

> The end-device switched to Class C but, didn't receive the downlinks.

I had similar issues . Make sure that gateway_ID in packet forwarder and ChirpStack Application Server (in Gateway configuration tab) are the same.

darkfader, 2019-12-06:
I was trying FUOTA as well and got:

```
[76831.401] ###### ===== Switch to Class C done.  ===== ######
```
then some class C downlink frame...
and then 
```
[76839.864] ###### =========== FRAG_DECODER ============ ######
[76839.864] ######               FINISHED                ######
[76839.865] ###### ===================================== ######
[76839.865] STATUS      : -2
[76839.865] CRC         : 00000000
```
It will receive several more downlink frames, but doesn't handle them because the state it FRAG_SESSION_NOT_STARTED.

It also doesn't switch back to class A (and set IsMcSessionStarted to false).

sathishkumarparthiban, 2019-12-06:
@KaDw 
On the gateway when i execute loraserver.js it listens for all the topics  and filters for the msgs

i get the Clocksync on the topic "join"
Something like "application/1/device/badcfe0000000051/join"
So i edited the loraserver.js to also listen for this msg.

// only interested in rx for now...
   if (!((/\/rx$/.test(topic))||(/\/join$/.test(topic)))) return;
After this i can see the msg flow , but i dont get the fPort parameters in the msg.

msg {"applicationID":"1","applicationName":"STM_BOARD","deviceName":"STM_BOARD_DEV","devEUI":"badcfe0000000051","devAddr":"01eaeffd","rxInfo":[{"gatewayID":"333833304bffffffff","uplinkID":"6ab0042a-9a51-4a9a-8001-3ff8e72cf815","name":"MY_CHN_GW_001","rssi":-103,"loRaSNR":8,"location":{"latitude":0,"longitude":0,"altitude":0}}],"txInfo":{"frequency":865402500,"dr":0}}
topic application/1/device/badcfe0000000051/join
Rx application/1/device/badcfe0000000051/join {
  applicationID: '1',
  applicationName: 'STM_BOARD',
  deviceName: 'STM_BOARD_DEV',
  devEUI: 'badcfe0000000051',
  devAddr: '01eaeffd',
  rxInfo: [
    {
      gatewayID: '333833304bffffffff',
      uplinkID: '6ab0042a-9a51-4a9a-8001-3ff8e72cf815',
      name: 'MY_CHN_GW_001',
      rssi: -103,
      loRaSNR: 8,
      location: [Object]
    }
  ],
  txInfo: { frequency: 865402500, dr: 0 }
}
device found
port is
undefined

msg {"applicationID":"1","applicationName":"STM_BOARD","deviceName":"STM_BOARD_DEV","devEUI":"badcfe0000000051","rxInfo":[{"gatewayID":"333833304bffffffff","uplinkID":"6ad49160-5a09-48d7-8c84-a56674db193f","name":"MY_CHN_GW_001","rssi":-113,"loRaSNR":5.8,"location":{"latitude":0,"longitude":0,"altitude":0}}],"txInfo":{"frequency":865402500,"dr":0},"adr":true,"fCnt":28,"fPort":1,"data":"FQ=="}
topic application/1/device/badcfe0000000051/rx
Rx application/1/device/badcfe0000000051/rx {
  applicationID: '1',
  applicationName: 'STM_BOARD',
  deviceName: 'STM_BOARD_DEV',
  devEUI: 'badcfe0000000051',
  rxInfo: [
    {
      gatewayID: '333833304bffffffff',
      uplinkID: '6ad49160-5a09-48d7-8c84-a56674db193f',
      name: 'MY_CHN_GW_001',
      rssi: -113,
      loRaSNR: 5.8,
      location: [Object]
    }
  ],
  txInfo: { frequency: 865402500, dr: 0 },
  adr: true,
  fCnt: 28,
  fPort: 1,
  data: 'FQ=='
}
device found
port is
1

KaDw, 2019-12-06:
@sathishkumarparthiban Sorry can't help you with loraserver.js script. Newest versions of loraserver (I think 3.0+) supports FUOTA from user interface, and this is what I'm using.

sathishkumarparthiban, 2019-12-06:
@KaDw thats fine.. How do you debug the msg flow on the gateway with user interface..

The problem i see is that my gateway isn't responding back for the device time sync. I'm able find this when i debugged the code from the script... 
Is there a way for the user interface to have a verbose log or something similar to see where its getting stuck??

KaDw, 2019-12-06:
@sathishkumarparthiban Check logs on the gateway, https://www.chirpstack.io/application-server/install/debian/#chirpstack-application-server-log-output

mluis1, 2020-03-23:
Just to inform that I was able to successfully run the fuota-test-01 with ChirpStack.

The interoperability test specification provided by the LoRa-Alliance uses a 1KB file which is split in 21 fragments of 50 bytes with a redundancy of 5 fragments.
The fuota-test-01 example is setup with these settings.
The below link provides a zip containing the binary test file.
[fuota-test-01-original-file.zip](https://github.com/Lora-net/LoRaMac-node/files/4370418/fuota-test-01-original-file.zip)

The ChirpSatck server on the other hand takes the 1KB file and splits it into fragments of maximum payload size for the used datarate.
In case a DR5 (SF27BW125) is used to download the file it means that one must change `FRAG_MAX_SIZE` definition on `FragDecoder.h` file from `50` to `255`

Run the test using ChirpStak:

1. Modify `FragDecoder.h`
2. Compile and flash the `fuota-test-01` example.
3. Power off the end-device
4. On ChirpStack server "CREATE FIRMWARE UPDATE JOB"
[image]
5. Power on the end-device
6. wait up until the process finishes.

**Note** I used the code provided on develop branch (commit b4b37db3d50111beacc040d080551af3586ec44b) and just changed the `FragDecoder.h` file.  
The used platform was a NucleoL476 and an SX1261MBXBAS shield.

Below the end-device serial terminal output

<details>
  <summary>Click to expand!</summary>
<p>

```log
###### ===================================== ######

Application name   : fuota-test-01
Application version: 1.0.0
GitHub base version: 4.4.3

###### ===================================== ######

DevEui      : 09-50-32-59-24-50-6C-59
AppEui      : 00-00-00-00-00-00-00-00
AppKey      : 2B 7E 15 16 28 AE D2 A6 AB F7 15 88 09 CF 4F 3C

###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK

###### =========== MLME-Confirm ============ ######
STATUS      : OK
###### ===========   JOINED     ============ ######

OTAA

DevAddr     :  0052798E

DATA RATE   : DR_3

###### =========== MLME-Request ============ ######
######            MLME_DEVICE_TIME           ######
###### ===================================== ######
STATUS      : OK

###### =========== MCPS-Request ============ ######
######           MCPS_UNCONFIRMED            ######
###### ===================================== ######
STATUS      : OK

###### =========== MCPS-Confirm ============ ######
STATUS      : OK

[... 731 lines cut ...]
###### ===================================== ######
STATUS      : OK

###### =========== MCPS-Confirm ============ ######
STATUS      : OK

###### =====   UPLINK FRAME       15   ===== ######

CLASS       : A

TX PORT     : 201
TX DATA     : UNCONFIRMED
05 18 A9 B2 EC

DATA RATE   : DR_5
U/L FREQ    : 868100000
TX POWER    : 7
CHANNEL MASK: 00FF
```
</p>
</details>

Below the final status provided by ChirpStack.
[image]
[image]

closed 2020-04-20 by mluis1

MarkoRimac, 2020-07-23:
Hello, has anyone tried FUOTA process using Chirpstack server and fuota-test-01 example lately? I've been trying to set it up and test it but have come to an issue I can't overcome.
Im using LmHelper files provided by semtech that implement FUOTA steps (Multicast, clock sync and fragmentation) but seem to have an issue of having different MC_NWK_S_KEY and MC_APP_S_KEY being generated on end-device side and on application server side. I've provided the same APP_KEY(GEN_APP_KEY in my case, cause lorawan version 1.0.3) to both end device and server side which is needed to obtain two MC keys needed, as it states in [lorawan multicast specification](https://lora-alliance.org/sites/default/files/2018-09/remote_multicast_setup_v1.0.0.pdf). I've set up LORAWAN version to 1.0.3 in application server device configuration because it says so in commissioning.h file that came with the semtechs lib implementing lora stack. 
Did anyone else come across the same issue or at least can confirm that the example is working with Chirpstack server?