Lora-net/LoRaMac-node #829: Need help config gateway CISCO.

VitorLima, 2019-11-25:
Hi guys, I have a client using a gateway CISCO and I am having a hard time to configure it.
First problem is it only opens the first 8 channels for up-link. Second is that it claims to use AU915 region configuration but it's in fact closer to US915. 
As I understood up-link goes from 902300000 to 903700000 (increments of +200000), BW_125KHZ, SF12. 
They sent me a channels.ini file with it's configuration. Can anyone help me setup my nodes configurations to work on this guy? Thanks and best regards.

`;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
;	LRC/LRR channels configuration
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;

;
; on uplink packet LRRs search the channel number [0..7] based on
; freqhz+modulation+bandwidth+datarate
;
; channel number, subband, datarate and coderate are transmitted to LRCs which
; can control subband duty-cycle limitation; LRCs also store channel number and
; datarate, coderate for downlink packets.
;
; for each channel it is possible to adjust the tx power against [gen].power
; in lgw.ini file or to set an absolute value
; 	[channel].power=+x	// power will be [gen].power + x
; 	[channel].power=-x	// power will be [gen].power - x
; 	[channel].power=x	// power will be x
;
; one channel is selected as the channel to use for RX2 window and it is 
; possible to set a downlink spreading factor
; 	by default this channel is LCd0/SF12
;

;;;;;;;;;;;;;;;;;;;;;;;;;;    US 915   ;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
;
; upstream LC0  .. LC63 [0..63] [902300000..914900000] +200000 BW_125KHZ
; upstream LC64 .. LC71 [64.71] [903000000..914200000] +1600000 BW_500KHZ
;
; downstream LCd0..LCd7 [127..134] [923300000..927500000] +600000 BW_500KHZ

;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
[channel:0]
	name=LC0
	subband=20
	freqhz=902300000
	bandwidth=${BW_125KHZ}
[channel:1]
	name=LC1
	subband=20
	freqhz=902500000
	bandwidth=${BW_125KHZ}
[channel:2]
	name=LC2
	subband=20
	freqhz=902700000
	bandwidth=${BW_125KHZ}
[channel:3]
	name=LC3
	subband=20
	freqhz=902900000
	bandwidth=${BW_125KHZ}
[channel:4]
	name=LC4
	subband=20
	freqhz=903100000
	bandwidth=${BW_125KHZ}
[channel:5]
	name=LC5
	subband=20
	freqhz=903300000
	bandwidth=${BW_125KHZ}
[channel:6]
	name=LC6
	subband=20
	freqhz=903500000
	bandwidth=${BW_125KHZ}
[channel:7]
	name=LC7
	subband=20
	freqhz=903700000
	bandwidth=${BW_125KHZ}
[channel:8]
	name=LC8
	subband=20
	freqhz=903900000
	bandwidth=${BW_125KHZ}
[channel:9]
	name=LC9
	subband=20
	freqhz=904100000
	bandwidth=${BW_125KHZ}
[channel:10]
	name=LC10
	subband=20
	freqhz=904300000
	bandwidth=${BW_125KHZ}
[channel:11]
	name=LC11
	subband=20
	freqhz=904500000
	bandwidth=${BW_125KHZ}
[channel:12]
	name=LC12
	subband=20
	freqhz=904700000
	bandwidth=${BW_125KHZ}
[channel:13]
	name=LC13
	subband=20
	freqhz=904900000
	bandwidth=${BW_125KHZ}
[channel:14]
	name=LC14
	subband=20
	freqhz=905100000
	bandwidth=${BW_125KHZ}
[channel:15]
	name=LC15
	subband=20
	freqhz=905300000
	bandwidth=${BW_125KHZ}
[channel:16]
	name=LC16
	subband=20
	freqhz=905500000
	bandwidth=${BW_125KHZ}
[channel:17]
	name=LC17
	subband=20
	freqhz=905700000
	bandwidth=${BW_125KHZ}
[channel:18]
	name=LC18
	subband=20
	freqhz=905900000
	bandwidth=${BW_125KHZ}
[channel:19]
	name=LC19
	subband=20
	freqhz=906100000
	bandwidth=${BW_125KHZ}
[channel:20]
	name=LC20
	subband=20
	freqhz=906300000
	bandwidth=${BW_125KHZ}
[channel:21]
	name=LC21
	subband=20
	freqhz=906500000
	bandwidth=${BW_125KHZ}
[channel:22]
	name=LC22
	subband=20
	freqhz=906700000
	bandwidth=${BW_125KHZ}
[channel:23]
	name=LC23
	subband=20
	freqhz=906900000
	bandwidth=${BW_125KHZ}
[channel:24]
	name=LC24
	subband=20
	freqhz=907100000
	bandwidth=${BW_125KHZ}
[channel:25]
	name=LC25
	subband=20
	freqhz=907300000
	bandwidth=${BW_125KHZ}
[channel:26]
	name=LC26
	subband=20
	freqhz=907500000
	bandwidth=${BW_125KHZ}
[channel:27]
	name=LC27
	subband=20
	freqhz=907700000
	bandwidth=${BW_125KHZ}
[channel:28]
	name=LC28
	subband=20
	freqhz=907900000
	bandwidth=${BW_125KHZ}
[channel:29]
	name=LC29
	subband=20
	freqhz=908100000
	bandwidth=${BW_125KHZ}
[channel:30]
	name=LC30
	subband=20
	freqhz=908300000
	bandwidth=${BW_125KHZ}
[channel:31]
	name=LC31
	subband=20
	freqhz=908500000
	bandwidth=${BW_125KHZ}
[channel:32]
	name=LC32
	subband=20
	freqhz=908700000
	bandwidth=${BW_125KHZ}
[channel:33]
	name=LC33
	subband=20
	freqhz=908900000
	bandwidth=${BW_125KHZ}
[channel:34]
	name=LC34
	subband=20
	freqhz=909100000
	bandwidth=${BW_125KHZ}
[channel:35]
	name=LC35
	subband=20
	freqhz=909300000
	bandwidth=${BW_125KHZ}
[channel:36]
	name=LC36
	subband=20
	freqhz=909500000
	bandwidth=${BW_125KHZ}
[channel:37]
	name=LC37
	subband=20
	freqhz=909700000
	bandwidth=${BW_125KHZ}
[channel:38]
	name=LC38
	subband=20
	freqhz=909900000
	bandwidth=${BW_125KHZ}
[channel:39]
	name=LC39
	subband=20
	freqhz=910100000
	bandwidth=${BW_125KHZ}
[channel:40]
	name=LC40
	subband=20
	freqhz=910300000
	bandwidth=${BW_125KHZ}
[channel:41]
	name=LC41
	subband=20
	freqhz=910500000
	bandwidth=${BW_125KHZ}
[channel:42]
	name=LC42
	subband=20
	freqhz=910700000
	bandwidth=${BW_125KHZ}
[channel:43]
	name=LC43
	subband=20
	freqhz=910900000
	bandwidth=${BW_125KHZ}
[channel:44]
	name=LC44
	subband=20
	freqhz=911100000
	bandwidth=${BW_125KHZ}
[channel:45]
	name=LC45
	subband=20
	freqhz=911300000
	bandwidth=${BW_125KHZ}
[channel:46]
	name=LC46
	subband=20
	freqhz=911500000
	bandwidth=${BW_125KHZ}
[channel:47]
	name=LC47
	subband=20
	freqhz=911700000
	bandwidth=${BW_125KHZ}
[channel:48]
	name=LC48
	subband=20
	freqhz=911900000
	bandwidth=${BW_125KHZ}
[channel:49]
	name=LC49
	subband=20
	freqhz=912100000
	bandwidth=${BW_125KHZ}
[channel:50]
	name=LC50
	subband=20
	freqhz=912300000
	bandwidth=${BW_125KHZ}
[channel:51]
	name=LC51
	subband=20
	freqhz=912500000
	bandwidth=${BW_125KHZ}
[channel:52]
	name=LC52
	subband=20
	freqhz=912700000
	bandwidth=${BW_125KHZ}
[channel:53]
	name=LC53
	subband=20
	freqhz=912900000
	bandwidth=${BW_125KHZ}
[channel:54]
	name=LC54
	subband=20
	freqhz=913100000
	bandwidth=${BW_125KHZ}
[channel:55]
	name=LC55
	subband=20
	freqhz=913300000
	bandwidth=${BW_125KHZ}
[channel:56]
	name=LC56
	subband=20
	freqhz=913500000
	bandwidth=${BW_125KHZ}
[channel:57]
	name=LC57
	subband=20
	freqhz=913700000
	bandwidth=${BW_125KHZ}
[channel:58]
	name=LC58
	subband=20
	freqhz=913900000
	bandwidth=${BW_125KHZ}
[channel:59]
	name=LC59
	subband=20
	freqhz=914100000
	bandwidth=${BW_125KHZ}
[channel:60]
	name=LC60
	subband=20
	freqhz=914300000
	bandwidth=${BW_125KHZ}
[channel:61]
	name=LC61
	subband=20
	freqhz=914500000
	bandwidth=${BW_125KHZ}
[channel:62]
	name=LC62
	subband=20
	freqhz=914700000
	bandwidth=${BW_125KHZ}
[channel:63]
	name=LC63
	subband=20
	freqhz=914900000
	bandwidth=${BW_125KHZ}

;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;

[channel:64]
	name=LC64
	subband=30
	freqhz=903000000
	bandwidth=${BW_500KHZ}
[channel:65]
	name=LC65
	subband=30
	freqhz=904600000
	bandwidth=${BW_500KHZ}
[channel:66]
	name=LC66
	subband=30
	freqhz=906200000
	bandwidth=${BW_500KHZ}
[channel:67]
	name=LC67
	subband=30
	freqhz=907800000
	bandwidth=${BW_500KHZ}
[channel:68]
	name=LC68
	subband=30
	freqhz=909400000
	bandwidth=${BW_500KHZ}
[channel:69]
	name=LC69
	subband=30
	freqhz=911000000
	bandwidth=${BW_500KHZ}
[channel:70]
	name=LC70
	subband=30
	freqhz=912600000
	bandwidth=${BW_500KHZ}
[channel:71]
	name=LC71
	subband=30
	freqhz=914200000
	bandwidth=${BW_500KHZ}

;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;

[channel:127]
	name=LCd0
	subband=40
	freqhz=923300000
	bandwidth=${BW_500KHZ}
	usedforrx2=1
	dataraterx2=${DR_LORA_SF12}
	power=+0
[channel:128]
	name=LCd1
	subband=40
	freqhz=923900000
	bandwidth=${BW_500KHZ}
[channel:129]
	name=LCd2
	subband=40
	freqhz=924500000
	bandwidth=${BW_500KHZ}
[channel:130]
	name=LCd3
	subband=40
	freqhz=925100000
	bandwidth=${BW_500KHZ}
[channel:131]
	name=LCd4
	subband=40
	freqhz=925700000
	bandwidth=${BW_500KHZ}
[channel:132]
	name=LCd5
	subband=40
	freqhz=926300000
	bandwidth=${BW_500KHZ}
[channel:133]
	name=LCd6
	subband=40
	freqhz=926900000
	bandwidth=${BW_500KHZ}
[channel:134]
	name=LCd7
	subband=40
	freqhz=927500000
	bandwidth=${BW_500KHZ}`

mluis1, 2019-11-27:
This question is not related to this project repository.
This repository concerns only end-devices.

We recommend that you ask your question to your gateway provider.

closed 2019-11-27 by mluis1

VitorLima, 2019-11-27:
Understood. I just wanted to understand better how to restrict channels and configure the region file.