Lora-net/LoRaMac-node #831: Errors happen when I added ADC on the SPI2 port

liuming15, 2019-12-06:
Hello everyone. I am going to implement the LoRaMac-Node on my project. My board combinate the STM32L073 Nucleo+SX1272MB2DAS and an ADC7799 which ADC7799 is for data-sampling and SX1272 will send the data to the PC. 
Now LoRaMac works for my board, so I want to add my AD7799 code to the project. But when I initialized the SPI2 in board.c using "SpiInit( &SPI2, SPI_2, AD7799_MOSI, AD7799_MISO, AD7799_SCLK, AD7799_NSS );". There is an error show:'..\board.c(169): error:  #158: expression must be an lvalue or a function designator
  				SpiInit( &
PI2, SPI_2, AD7799_MOSI, AD7799_MISO, AD7799_SCLK, AD7799_NSS );'

The SPI2 is defined as '#define SPI2                ((SPI_TypeDef *) SPI2_BASE)' in stm32l073xx.h 

So what's wrong with me? Any help will be appreciated.

mluis1, 2020-02-21:
Have you solved your issue?
Can we close the issue?

mluis1, 2020-03-24:
No feedback. We close the issue

closed 2020-03-24 by mluis1