Lora-net/LoRaMac-node #832: Stack stucks in LoRaMacIsBusy() in battery mode

AliceWribbit, 2019-12-08:
Hi!
this issue was created upon @djaeckle request in #757 issue
If Rx error occurs during reception of packet in RX1 the stack stucks in LORAMAC_TX_RUNNING state.
I actually faced the same problem that Stack stucks in LORAMAC_TX_RUNNING,and function LoRaMacIsBusy() will always return busy and can't send data anymore. I fix the problem with the code in #757 issue but only in USB charging mode. If i switch to battery mode, the problem happened again.It seemed that the problem haven't solved in low power mode.
In usb charging mode (bool UsbIsConnected = true;),i test it for 7days, no problem. In battery mode(bool UsbIsConnected = false;),the problem appears easily especially when cutting off internet access of gateway after node OTAA joined.
My hardware playform is NucleoL073 with SX1278.

AliceWribbit, 2019-12-08:
release version both V4.4.2 and V4.4.1 have this problem.

mluis1, 2020-03-18:
In version [v4.4.3](https://github.com/Lora-net/LoRaMac-node/releases/tag/v4.4.3) we provided a fix for issue #767 which I think should solve the issue that you are observing.
The changes are located at commit 35bda48bdcf3a2237045843c095b007c911e93eb
Could you please check if it solves the observed issue?

mluis1, 2020-04-03:
No feedback, we close the issue.

closed 2020-04-03 by mluis1

alexrayne, 2020-12-21:
Obviously this is not a solution. It is a trash.