Lora-net/LoRaMac-node #834: How do you correctly request a rejoin

ChrisEAlfred, 2019-12-15:
If you wanted to force a rejoin form an asynchronous event (e.g. on button press or timer), what is the correct way to force a rejoin in the main loop (class A main.c from examples)?

mluis1, 2019-12-16:
You can just do a `JoinNetwork` function call.

The certification protocol handling does this when `0x06` command is sent to the `FPort`  224.
In this case everything except the `JoinNetwork` function call concerns the certification protocol handling.

```c
                case 6: // (ix)
                    {
                        // Disable TestMode and revert back to normal operation
                        IsTxConfirmed = LORAWAN_CONFIRMED_MSG_ON;
                        AppPort = LORAWAN_APP_PORT;
                        AppDataSize = AppDataSizeBackup;
                        ComplianceTest.DownLinkCounter = 0;
                        ComplianceTest.Running = false;

                        MibRequestConfirm_t mibReq;
                        mibReq.Type = MIB_ADR;
                        mibReq.Param.AdrEnable = LORAWAN_ADR_ON;
                        LoRaMacMibSetRequestConfirm( &mibReq );
#if defined( REGION_EU868 ) || defined( REGION_RU864 ) || defined( REGION_CN779 ) || defined( REGION_EU433 )
                        LoRaMacTestSetDutyCycleOn( LORAWAN_DUTYCYCLE_ON );
#endif

                        JoinNetwork( );
                    }
                    break;
```

ChrisEAlfred, 2019-12-16:
But don't you also need to set MIB_NETWORK_ACTIVATION back to ACTIVATION_TYPE_NONE and possibly set the device state to DEVICE_STATE_JOIN?

Otherwise (I think) the main loop state machine will not stay in 'joining' if the one JoinNetwork call does not successfully join.

mluis1, 2019-12-16:
No, there is no need. Just the call to  `JoinNetwork` is necessary.

For instance the end-device could be initially setup to use ABP activation and then one could trig a command to force the end-device join by OTAA. The only thing to do is to call `JoinNetwork` function.
This is what is done on the certification protocol handling.

There is no need to set the `DEVICE_STATE_JOIN` state as each time the `OnTxNextPacketTimerEvent` callback is executed and the activation didn't succeed the `JoinNetwork` function is automatically called again.

The states `DEVICE_STATE_START` and `DEVICE_STATE_JOIN` are called only once when the end-device boots and in the case the `NvmCtxMgmtRestore` function fails.

```c
static void OnTxNextPacketTimerEvent( void* context )
{
    MibRequestConfirm_t mibReq;
    LoRaMacStatus_t status;

    TimerStop( &TxNextPacketTimer );

    mibReq.Type = MIB_NETWORK_ACTIVATION;
    status = LoRaMacMibGetRequestConfirm( &mibReq );

    if( status == LORAMAC_STATUS_OK )
    {
        if( mibReq.Param.NetworkActivation == ACTIVATION_TYPE_NONE )
        {
            // Network not joined yet. Try to join again
            JoinNetwork( ); // <-- Join network again.
        }
        else
        {
            DeviceState = DEVICE_STATE_SEND;
            NextTx = true;
        }
    }
}
```

Please also note that the activation type is reset every time the `mlmeRequest->Type = MLME_JOIN`.

```c
    switch( mlmeRequest->Type )
    {
        case MLME_JOIN:
        {
            if( ( MacCtx.MacState & LORAMAC_TX_DELAYED ) == LORAMAC_TX_DELAYED )
            {
                return LORAMAC_STATUS_BUSY;
            }

            ResetMacParameters( ); // <-- Activation type reset to `ACTIVATION_TYPE_NONE `

            MacCtx.NvmCtx->MacParams.ChannelsDatarate = RegionAlternateDr( MacCtx.NvmCtx->Region, mlmeRequest->Req.Join.Datarate, ALTERNATE_DR );

            queueElement.Status = LORAMAC_EVENT_INFO_STATUS_JOIN_FAIL;

            status = SendReJoinReq( JOIN_REQ );

            if( status != LORAMAC_STATUS_OK )
            {
                // Revert back the previous datarate ( mainly used for US915 like regions )
                MacCtx.NvmCtx->MacParams.ChannelsDatarate = RegionAlternateDr( MacCtx.NvmCtx->Region, mlmeRequest->Req.Join.Datarate, ALTERNATE_DR_RESTORE );
            }
            break;
        }
```

mluis1, 2020-02-21:
Can we close this issue?

mluis1, 2020-03-24:
No feedback. We close the issue

closed 2020-03-24 by mluis1