Lora-net/LoRaMac-node #835: Need for clarification about Stack versions

Adamzat, 2019-12-19:
Hi every one,
We are using the LoraWan Stack version 4.4.0 (spec 1.0.2) for our devices (Class A). Now we are thinking to move to version 4.4.3 (spec 1.0.3). To do so, we need clarification about the questions below:
- In the source code of version 4.4.3, I see in the Mib Structure that many new implemented keys that belong to LoraWan spec.1.1 and not only 1.0.3. That makes some confusion for us. Can you please clarify if this version supports both 1.0.3 and 1.1?
- Do you have an idea when the Stack supporting spec 1.0.4 and spec 1.1 will be released (months, year) ?
Thank you.
Best regards.

Pirakutu, 2019-12-19:
Hi,

I would also like to double on the second point you mentioned. Do you have any kind of rough roadmap for the new stack version supporting 1.0.4?

Cheers,

Andres

ConnyBusy, 2019-12-19:
-  LoRaWAN v1.0.3 was intended to work on a v1.1 and vice versa. But are points which weren't clear in the specification that made such an interoperability difficult. Hence, v1.0.4 which back ported most features of v1.1 making such interoperability possible.

- Stack for v1.0.4 is currently under development which shall be released as version 4.5.0. Timeline I am not sure.

Adamzat, 2019-12-20:
Hi ConnyBusy,
Thank you for your response. Just to be sure, regarding the interoperability aspect, you mean that future devices that supports the version v1.0.4 (for example) can communicate with LoraWan servers supporting v1.1 and vice versa ?
Thank you.

ConnyBusy, 2019-12-20:
Yes, that's the idea behind it. This will prevent Network servers having multiple versions of their servers. 
Of course, there will be an out-of-band communication between device manufacture/owner and network provider so that the server would be able to send only mac commands available in the version and or use the appropriate keys for security aspects.

Adamzat, 2019-12-20:
Thank you. It is clear.

closed 2019-12-20 by Adamzat