Lora-net/LoRaMac-node #842: SX1276 fails to receive large payloads

nils-grepit, 2020-01-22:
I seem to be unable to receive large payloads, similar to the reports in #622.

Running the ping-pong example on `develop` (87f19e84ae2fc4af72af9567fe722386de6ce9f4) with 2 B-L072Z-LRWAN1, I see LD1 blink as expected. If I modify the example to use a larger payload, e.g.
```
#define BUFFER_SIZE                                 70 // Define the payload size here
```
I do not see any activity on LD1. This appears to be because the packets are sent, but not received.

mluis1, 2020-02-26:
This issue should only be observed when SX1272 or SX176 radios are used.

While we don't provide an update on the respective drivers could you please apply the below patch.

The update of the radio drivers will take some more time as we need to validate that the required changes don't break something else.

```patch
diff --git a/src/apps/ping-pong/B-L072Z-LRWAN1/main.c b/src/apps/ping-pong/B-L072Z-LRWAN1/main.c
index 6c11da8b..27ca9276 100644
--- a/src/apps/ping-pong/B-L072Z-LRWAN1/main.c
+++ b/src/apps/ping-pong/B-L072Z-LRWAN1/main.c
@@ -213,6 +213,8 @@ int main( void )
     #error "Please define a frequency band in the compiler options."
 #endif
 
+    Radio.SetMaxPayloadLength( MODEM_LORA, BUFFER_SIZE );
+
     Radio.Rx( RX_TIMEOUT_VALUE );
 
     while( 1 )
```

tmplt, 2020-03-02:
Issue reproduced in fbb6a97774e63643a11e4d3334458efd7879867b. Above patch resolved the issue and works with a buffer size up to 255.

closed 2020-03-10 by mluis1