Lora-net/LoRaMac-node #845: Node not able to receive JoinAccept from server

Mani-Sadhasivam, 2020-02-15:
Hello,

I'm working on adding LoRaWAN support to Zephyr RTOS based on LoRaMac-node repo. Right now I'm focusing on getting ClassA support working. In this, I can successfully send `Join Req` to TTN server and the server sends the `Join Accept` in response to that (confirmed by looking into the console). But I can't receive the `Join Accept` message on node and always getting Rx Timeout in both windows. I tried modifying the Rx delay, Symbol timeout etc... but nothing helped. However, if I use the vendor specific firmware based on AT commands, I can receive the `Join Accept` message on node, so this confirms that the gateway has no issue and there could be something wrong with my code in Zephyr.

It would be really helpful if someone can shed some light here.

The WIP code can be found below:
https://github.com/Mani-Sadhasivam/zephyr/blob/lorawan-temp/subsys/lorawan/lorawan.c
https://github.com/Mani-Sadhasivam/zephyr/blob/lorawan-temp/samples/lorawan/src/main.c

**Note** The code is still WIP, so there are several dirty hacks just for proof of concept.

mluis1, 2020-02-18:
Glad to see this work being done. Thank you.

Most of the time this kind of issues are related to timing issues. I didn't had time to deeply look into the port being done but, I can provide some ideas on how to figure out what is going on.

Things to check:
  * Validate radio driver operation when performing a point to point communication. Ping-Pong kind of communication.
  * Validate that the IQ inversion is correctly handled. Tx non inverted, Rx inverted.
  * Increase the MAC layer tolerance to timing errors by changing the `SystemMaxRxError` parameter
   ```c
                mibReq.Type = MIB_SYSTEM_MAX_RX_ERROR;
                mibReq.Param.SystemMaxRxError = 20;
                LoRaMacMibSetRequestConfirm( &mibReq );
   ```
   Please note that we cannot increase this to values bigger than 30 milliseconds. If I remember well we start to see other side effects when the value is bigger. By default the value is set to 10 milliseconds in the MAC layer. Changing it to 20 milliseconds is a good compromise.
  * In order to debug timing issues the best thing to do is to use 2 debug pins on the MCU and an oscilloscope.  
    Please take a look at the following code on radio driver.  
https://github.com/Lora-net/LoRaMac-node/blob/87f19e84ae2fc4af72af9567fe722386de6ce9f4/src/radio/sx1276/sx1276.c#L1187
   Then if you have access to the gateway antenna connector you can connect it directly to an oscilloscope input set for 50 Ohms impedance. (Please note that this signal will be quite small. a few millivolts), You may use an RF splitter in order to connect the cable as well as an antenna. By doing this you will be able to see when the gateway transmits.
   You can setup the oscilloscope trigger on falling edge of Tx debug pin signal.
   You should then be able to observe something similar to the image below.
[image]

Hope these hints help and good luck.

Mani-Sadhasivam, 2020-02-20:
@mluis1 Thanks a lot for the info! Unfortunately, I don't have an oscilloscope for RF. But I have done some investigation and found that the `DIO1` (Rx Timeout) pin gets asserted always during Rx Window indicating timeout. Even if I try to receive using `Radio.Rx(0)` (just for testing), I don't get any `DIO0` Irq. Quite surprisingly, register settings of SX1276 seem to be sane but still not receiving no data even though Gateway is transmitting.

Mani-Sadhasivam, 2020-02-21:
@mluis1 Got it working now. It turned out that I had to adjust Rx window delays (many thanks to Kwon Tae Young for spotting this). I did try to modify the Rx window delays before but could've missed something.

Thanks for your help!

closed 2020-02-21 by Mani-Sadhasivam

mluis1, 2020-02-21:
Nice, well done.

davidatingenuity, 2020-05-15:
> Please note that we cannot increase this to values bigger than 30 milliseconds. If I remember well we start to see other side effects when the value is bigger. By default the value is set to 10 milliseconds in the MAC layer. Changing it to 20 milliseconds is a good compromise.

Can you provide any more information about these side effects? I increased our MIB_SYSTEM_MAX_RX_ERROR to compensate for a low-accuracy timing source, and I started consistently missing the first 4 ACKs on confirmed messages. Is there anything we can do to make sure we hit the rx windows with a low-accuracy time source, and still reliably receive ACKs?