Lora-net/LoRaMac-node #846: How to trim crystal oscillator of sx1261?

Osamusii, 2020-02-20:
HI,

I'm trying to trim crystal oscillator frequency, and read Application Note: Crystal Oscillator Trimming(AN1200.33). But just after function `Radio.SetTxContinuousWave( RF_FREQUENCY, TX_OUTPUT_POWER, TX_TIMEOUT );`, the XTA&XTB internal cap. register is set to default value. The TX frequency is not changed.
I'm not able to understand why the register is set to default. 
Here is code:
```
int main( void )
{
    // Target board initialization
    BoardInitMcu( );
    BoardInitPeriph( );

    TimerInit( &Led1Timer, OnLed1TimerEvent );
    TimerSetValue( &Led1Timer, 90 );

    TimerInit( &Led2Timer, OnLed2TimerEvent );
    TimerSetValue( &Led2Timer, 90 );

    // Switch LED 1 ON
    GpioWrite( &Led1, 0 );
    TimerStart( &Led1Timer );

    // Radio initialization
    RadioEvents.TxTimeout = OnRadioTxTimeout;
    Radio.Init( &RadioEvents );
    
    	 ///* WIP: XTAL oscillator trimming /////
		 SX126xSetStandby( STDBY_XOSC );							//sx126x.h
		 Radio.Write( REG_XTA_TRIM, 0x01 );
		 Radio.Write( REG_XTB_TRIM, 0x01 );
		 int xta = Radio.Read( REG_XTA_TRIM );
		 int xtb = Radio.Read( REG_XTB_TRIM );
		 int mode = SX126xGetOperatingMode();
		 printf( "REG_XTA,B_TRIM = %#x, %#x\n\rmode = %d\n\r" , xta , xtb , mode );
		 
    Radio.SetTxContinuousWave( RF_FREQUENCY, TX_OUTPUT_POWER, TX_TIMEOUT );

		 xta = Radio.Read( REG_XTA_TRIM );
		 xtb = Radio.Read( REG_XTB_TRIM );
		 mode = SX126xGetOperatingMode();
		 printf( "After TX:REG_XTA,B_TRIM = %#x, %#x\n\rmode = %d\n\r" , xta , xtb , mode );
		
    // Blink LEDs just to show some activity
    while( 1 )
    {...
```
Thank you in advance.

Osamusii, 2020-02-21:
I realized that following codes must be included SetTxContinuousWave() of radio.c:
```
SX126xSetStandby( STDBY_XOSC );							//sx126x.h
		 Radio.Write( REG_XTA_TRIM, 0x01 );
		 Radio.Write( REG_XTB_TRIM, 0x01 );
```
I could change the TX frequency! Thank you!

closed 2020-02-21 by Osamusii

mlegrand2, 2021-08-06:
Hello,
This is only for continuous wave mode... The function SetTxContinuous is used only for RF adjstements.
How to trim crystal in normal operation? Where do I have to insert the write to TXA and TXB_TRIM registers?
Thank you for your help.