Lora-net/LoRaMac-node #852: Wrong MacCtx.NvmCtx->Version structure initialization  ?

FranciscoRV, 2020-03-05:
https://github.com/Lora-net/LoRaMac-node/blob/15fa96bd7b1e6c6374ca5a7260eca5b69eba232d/src/mac/LoRaMac.c#L3169

**MacCtx.NvmCtx->Version value is set to :**
- Version.Fields.Major    = 1;
- Version.Fields.Minor    = 0;
- Version.Fields.Revision = 0;
- **Version.Fields.Patch    = 3;**

**It's should be :**
- Version.Fields.Major    = 1;
- Version.Fields.Minor    = 0;
- **Version.Fields.Revision = 3;**
- Version.Fields.Patch    = 0;

**By changing structure order this is fixed:**

https://github.com/Lora-net/LoRaMac-node/blob/86e177079e7c3da4a62e1ce1a14d13af71374cf2/src/boards/utilities.h#L77

```
typedef union Version_u
{
    struct Version_s
    {
        uint8_t Patch;
        uint8_t Revision;
        uint8_t Minor;
        uint8_t Major;
    }Fields;
    uint32_t Value;
}Version_t;
```

mluis1, 2020-03-05:
We think that the current implementation is correct.

We are trying to follow the [Semantic Versioning 2.0.0](https://semver.org) recommendations.

If you look at the `Version_t` definition
https://github.com/Lora-net/LoRaMac-node/blob/86e177079e7c3da4a62e1ce1a14d13af71374cf2/src/boards/utilities.h#L77
You will notice that we follow the recommendation

```c
/*!
 * Version
 */
typedef union Version_u
{
    struct Version_s
    {
        uint8_t Revision;
        uint8_t Patch;
        uint8_t Minor;
        uint8_t Major;
    }Fields;
    uint32_t Value;
}Version_t;
```
This means that LoRaWAN 1.0.3 -> `0x01000300` will be encoded as:
```
Major    : 01
Minor    : 00
Patch    : 03
Revision : 00
```

The Regional Parameters RP2-1.0.1 -> 0x02010001 will be encoded as:
```
Major    : 02
Minor    : 01
Patch    : 00
Revision : 01
```

FranciscoRV, 2020-03-05:
Thank you. It is clear 👍

closed 2020-03-05 by mluis1