Lora-net/LoRaMac-node #854: AES encryption abstraction layer

omogenot, 2020-03-07:
Hi Team,
This is not an issue, more a contribution proposal.
I wrote another version of the secure-element section that abstract the calls to the AES CBC 128 routine in such a way that a port from software to hardware AES engine can easily be integrated. On the top of it, the CMAC is generated using the same basic AES CBC128 routine, no need for another external reference.
In other words, the secure-element implementation is using only three "extern" calls that depend on the AES crypto implementation (hardware or software), without any reference to foreign data structures:
* AESStart() to initialise (if need be) the AES crypto driver (hardware or software)
* AESCBC128Bits(...) to perform encryption on blocks
* AESStop() to de-initialise (if need be) the AES crypto driver (hardware or software)

Any interest ?

mluis1, 2020-03-10:
Hi @omogenot,

All propositions are welcome.
When a proposition is received we will try to review it  and provide feedback as soon as possible.

If the proposition is inline with our requirements/expectations then we will gladly accept it.

mluis1, 2020-03-26:
Please refer to pull-request #855

closed 2020-03-26 by mluis1