Lora-net/LoRaMac-node #856: AU915 Always Enable 500Khz Channels

gilbertoe, 2020-03-11:
Hi

The Lorawan service provider only uses an 8 + 0 channel configuration (without 500khz channels).
I used the code below to configure this band:

static uint16_t GatewayChannelsMask [6] = {0xFF, 0x00, 0x00, 0x00, 0x00, 0x00};

                mibReq.Type = MIB_CHANNELS_DEFAULT_MASK;
                mibReq.Param.ChannelsDefaultMask = GatewayChannelsMask;
                LoRaMacMibSetRequestConfirm (& mibReq);

                mibReq.Type = MIB_CHANNELS_MASK;
                mibReq.Param.ChannelsMask =  GatewayChannelsMask;
                LoRaMacMibSetRequestConfirm (& mibReq);

I inserted the code in DEVICE_STATE_START state of the main program.

I get the following LOG when a uplink message is sent:

###### ===== UPLINK FRAME 4 ==== ######                                         
                                                                                
CLASS       : A                                                                 
                                                                                
TX PORT     : 2                                                                 
TX DATA     : UNCONFIRMED                                                       
00                                                                              
                                                                                
DATA RATE   : DR_2                                                              
U/L FREQ    : 916200000                                                         
TX POWER    : 0                                                                 
CHANNEL MASK: 00FF 0000 0000 0000 00FF    

It seems that 500Khz channels are always being enabled by the Lorawan stack.

Is this the correct way to apply the channels mask?

Please advise.

BR
Gilberto

EduardoPfeifer, 2020-03-11:
uint16_t channelMask[] = { 0x00FF, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000 };

> Hi
>
> The Lorawan service provider only uses an 8 + 0 channel configuration (without 500khz channels).
> I used the code below to configure this band:
>
> static uint16_t GatewayChannelsMask [6] = {0xFF, [...]

Try this:

uint16_t channelMask[] = { 0x00FF, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000 };

closed 2020-03-11 by gilbertoe

acutetech, 2020-06-15:
I wonder if this is because of the use of CHANNELS_MASK_500KHZ_MASK, as in these lines:

>     // Re-enable 500 kHz default channels
>     NvmCtx.ChannelsMask[4] = CHANNELS_MASK_500KHZ_MASK;

Perhaps try redefining CHANNELS_MASK_500KHZ_MASK (default is 0x00ff)